iNavFlight/inav #3049: Matek F405 CTR will not arm. All pre-checks green, all sensors good and GPS 3Dfix.

evmoius, 2018-04-15:
Matek F405 CTR
INAV/MATEKF405 1.8.0 Dec  7 2017 / 16:58:06 ()
Got this .HEX from the Matek Site: http://www.mateksys.com/?portfolio=f405-ctr#tab-id-7

My 800mm Hex X will not arm.  I've done everything recommended. 
All calibrations done
All pre-arm checks are green
All sensors are working properly under sensors and are displayed as blue
GPS has a 3d fix with 10+ sats
Have repeatedly verified that the arming switch is properly setup.

Have watched several vids and been through forums. I can't find any reason why this is going on so I'm posting here.

[PasteBin](https://pastebin.com/ExDhq2Zp)

digitalentity, 2018-04-15:
Please go to CLI and check output of `status` command. Look for `Arming disabled flags:` line - those are reasons for arming being blocked.

evmoius, 2018-04-16:
Thanks for responding @digitalentity.
I don't see any Arming disabled flags:

[https://pastebin.com/Jz8By3Yy](url)

evmoius, 2018-04-16:
Re-flashed the board again and now it arms just fine. Super weird.  I'm going to keep an eye on things.

closed 2018-04-16 by evmoius