iNavFlight/inav #3050: BB is not logging to SD card [OMNIBUSF4PRO]

artempolikarpov, 2018-04-15:
I have INAV/OMNIBUSF4PRO 1.9.0 and set BB logging device - SD card.
But when I reject and browse SD card - there is no log

digitalentity, 2018-04-19:
Try a different SD card. Not all SD cards work correctly in SPI mode

artempolikarpov, 2018-04-19:
standard 16gb, fat32, no warning messages in configurator

giacomo892, 2018-04-19:
Have you enabled logging from the configuration tab? Is the SDCARD recognized?

artempolikarpov, 2018-04-19:
> Have you enabled logging from the configuration tab? Is the SDCARD recognized? 
Sure - BlackBox enabled as feature, SDCARD is recognized in BlackBox tab

giacomo892, 2018-04-19:
Note that BB only works when copter is armed.

artempolikarpov, 2018-04-19:
I know, I made few flights, sdcard is still empty

DzikuVx, 2018-04-23:
Tested 1.9.1 on Omnibus F4 Pro v2 and all the logs were correctly recorded. So I assume hardware or configuration

closed 2018-05-16 by DzikuVx