iNavFlight/inav #3064: PIDs are reset to defaults after PID profile change via OSD

artempolikarpov, 2018-04-17:
**Prerequisites:**
FC with OSD like OMNIBUSF4PRO
PID profile 1 is selected, pids are differ from default

**Steps:**

1. Open OSD menu with sticks combination
2. Change PID profile to 2 and Save & Exit
3. Open OSD menu with sticks combination
4. Change PID profile to 1 and Save & Exit

**Actual result:**
PIDs value are reset to default

shellixyz, 2018-04-19:
Confirmed. I can reproduce on 1.9.0.

digitalentity, 2018-04-19:
@artempolikarpov I can't reproduce the issue on exact same setup. Can you please record a video (DVR recording to show OSD activity) to demonstrate the issue?

artempolikarpov, 2018-04-19:
I think I found the source of issue. But I'm not sure is it a bug or "as designed"

When you change PID profile and then open PIDs for edit - PIDs are displayed from previous PID profile - this could be the issue.

If you change PID profile and Save and Exit - pids will be changed after this action.
BUT if you change PID profile and then open PIDs and edit them, then Save and Exit action - your edited pids from previous pid profile will present.

https://drive.google.com/file/d/1rP51WvFcBQ6DrtMyoEeJFZb0lvAwswg-/view?usp=sharing

PR #3433 (2018-06-25): When editing PID profiles from CMS, change the profile immediately
This way we make sure we're always reading and writing to the
selected PID profile without having to rely on so much intermediate
data (with its assocciated RAM costs). It also fixes the problem
when data from a PID profile could accidentaly overwrite another.

Fixes #3064
commit fb6ff202 (2018-06-25): When editing PID profiles from CMS, change the profile immediately

This way we make sure we're always reading and writing to the
selected PID profile without having to rely on so much intermediate
data (with its assocciated RAM costs). It also fixes the problem
when data from a PID profile could accidentaly overwrite another.

Fixes #3064
closed 2018-06-25 by fiam