iNavFlight/inav #3071: Feature Request: I need more data on OSD

Denisfk1985, 2018-04-18:
I fly to a height of 8 or more km, and I would like to see on OSD the following values, which are already in the environment of variables, some are easily calculated:

1. Temperature from the barometer sensor. (We did it in custom, the minimum temperature, I was able to fix bmp-280 in my flight -53 degrees Celsius, at an altitude of 9 kilometers)
2. GPS altitude (In practice, it differs from the altitude using bmp280, especially at high altitude at extremely low temperatures due to the error of the sensor itself)
2. Efficiency of energy use in flight vertically up (climb), in watts or amperes per kilometer, for example, as it is done for horizontal flights.

Thanks.

wx4cb, 2018-04-18:
i believe GPS alt is already in the latest one, that was one of the things i wanted heh 

As far as efficiency, if it's already in use for horizontal flights wouldn't it be the same if you were vertical? i can't see how it would matter to the efficiency whether you're vertical or horizontal, (you wouldn't be going into a head wind for example)?

Denisfk1985, 2018-04-18:
> As far as efficiency, if it's already in use for horizontal flights wouldn't it be the same if you were vertical? >i can't see how it would matter to the efficiency whether you're vertical or horizontal, (you wouldn't be >going into a head wind for example)?

The fact is that I need to always find the optimal value of the throttle level. In horizontal flight, the efficiency ma / km is displayed in the horizontal plane, but I would like to have the same option for the vertical plane. for example, the current consumption is 20 amps, the lifting speed is 10 meters per second, on the basis of this, the lifting efficiency will be 20/10 = 2 amps per meter per second of lift.
For 12 amperes and 3 meters per second there will already be another efficiency of 12/3 = 4 amperes per meter per second.

wx4cb, 2018-04-18:
im not a physicist lol i understand that reasoning, but i don't see how it would be any different either vertically or horizontally. if you turn on "efficiency" in the osd then it wouldn't matter if you're horizontal or vertical i would have thought ?

alexeystn, 2018-04-18:
Efficiency is calculated here, based on _ground_ speed, as I understand:
https://github.com/iNavFlight/inav/blob/master/src/main/io/osd.c#L1469
If you fly forward and ascend at the same time, for example, 20 km/h forward + 20 km/h upwards, it will use 1 km/h instead of sqrt(2)=1.41.
That's an interesting idea, I think, to combine vertical speed with horizontal speed for distance and efficiency calculation.

closed 2018-07-01 by stale[bot]