iNavFlight/inav #3087: PID attenuation for VTOL

xswein, 2018-04-21:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

xswein, 2018-04-21:
CleanFlight issues seem to have some discussion about "Allow PID tuning with a TX POT" https://github.com/cleanflight/cleanflight/issues/10
Issues #10 #173 more recently #1842

taileron, 2018-06-14:
I´ve done it with Betaflight 3.22 but I don´t know how to integrate it into iNav or Cleanflight.
It bases of adding a float value that varies from 0 to 1 if an aux channel goes from 1ms to 2ms.
Added scaling factors into the final pid summing procedure.
float auxPgain = (((float)rcData[AUX5])/2000.0f)-0.5f;
float auxIgain = (0.5f+(2.0f * auxPgain))/1.5f;

So aux5 directly controls P and D term 0 .... 100% and iterm 33 .... 100%

taileron, 2018-06-14:
Another workaround can be done with combination of TPA and rate profiles that can be changed inflight
profile 1 with tpa rate = 0 and profile 2 with tpa rate = 70% starting at 0% throttle (1000us)
profile 1 can be used for hovering, profile 2 for transition and fw
that works for me perfect in a Hyper Taxi (eFlight) 
For using angle mode the mapping has to be changed from "TAER" to "TREA" yaw is used for roll and vice versa

xswein, 2018-06-15:
Thanks a lot, TPA profile solution is good idea, I'll try it this weekend!

DzikuVx, 2019-01-05:
On a good side, I'm working on conditional mixer that will be required for VTOL too

closed 2023-09-29 by DzikuVx