iNavFlight/inav #3094: Custom mixer add "Butterfly" option for electric Glider

advancetom, 2018-04-22:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

DzikuVx, 2018-04-22:
@advancetom Please cleanup ticket description. If you have a working smix, please provide

advancetom, 2018-04-22:
are flaps the same as butterfly dont understand you?

Redshifft, 2018-04-25:
Butterfly/Crow is generally a 4 servo wing braking - Flaps down and Ailerons up
Spoilerons is generally just Ailerons up as brakes.
A 4 servo glider wing mix also generally moves inboard ailerons 50% of outboard for roll control also
 can camber or reflex the full control surfaces.
All slightly different and I doubt understood by many around here :D come the revolution and the open smixer, when you have worked out what you actually want you should be able to do it.

closed 2018-07-08 by stale[bot]

jgilbertfpv, 2018-08-07:
I'm interested in this as well, I just got a fx-79 buffalo wing with split elevons i'd like to do crowe braking with. Is there a mix for it or is this dead and forgotten?
there's a flapperon mode to set right? well could it be inverted so elevons go up instead of down, and then set another two channels to go down instead of up as normal flapperon mode