iNavFlight/inav #3097: FPORT SPF3 1.9.1 Telemetry working but Receiver not

SPlissken, 2018-04-22:
I've just setup FPORT on an R-XSR using LBT Firmware 180228 this is connected to TX on UART 3 on an SPF3 clone running 1.9.1 (release).

Smartport was working perfectly bi-directional before the change to FPORT with the same connection: PIDs could be updated and Telemetry data was received.

Now the values are as follows and no hardware changes have been made at all:
serialrx_provider = FPORT
serialrx_inverted = ON
serialrx_halfduplex = ON
receiver_type = SERIAL

Telemetry is working fine and all data is displaying and updating on my QX7.

Weird thing is that the receiver tab doesn't show any data coming through just static 1500 for all apart from static 850 for throttle and also the RSSI shows as 0.

I've tried changing serialrx_inverted and serialrx_halfduplex  but all this does is stop telemetry coming through to the QX7.

I know that this could well be an RX problem with beta-firmware etc. but could it be an issue with iNav, has anyone else seen this?

SPlissken, 2018-04-23:
SOLVED --- R-XSR was in PPM mode

closed 2018-04-23 by SPlissken

teckel12, 2018-07-06:
@SPlissken What is the physical connection between your FC and the XSR.  Considering setting up F.Port but a little confused as to what port to use on the FC (an inverted port?) and how to connect to the XSR, which is inverted.

SPlissken, 2018-07-06:
@teckel12 I didn't change any wiring at all it was all done in configuration for me.
I'm using an F3 so I just used the existing S.Port to HW UART wiring and set
`serialrx_provider = FPORT`
`serialrx_inverted = ON`
`serialrx_halfduplex = ON`
`receiver_type = SERIAL`

I think that unless your are tapping the un-inverted signal from the R-XSR you should probably go ahead and use an inverted port as you would normally do for SBUS/S.Port.
My impression is that the developers/FrSky have made the F.Port firmware as much a drop in replacement as possible.

The only thing that caught me out was that after flashing, my R-XSR reset to PPM mode which meant that although it was putting out F.Port containing telemetry data the control data was missing. Switching PPM off fixed the issue immediately.

teckel12, 2018-07-06:
I tried using these settings with F.Port last night and I couldn't get the controller to work nor telemetry.

You're using F3 so there's not an on-FC inverter. I'm just very confused with what the settings do as far as inversion goes. I don't think they actually invert, only trigger an inversion on particular boards. I believe the only inversion INAV can do is on the soft serial port. Then there's the confusion with every F.Port thing I read that talks about wiring up the receiver to bypass inversion.

Anyway, I tried and it seems that there's more to it than just flashing the receiver and setting a few options. There need to be more detailed information to get this working.

SPlissken, 2018-07-06:
F3's have native inversion. If you have an on-board inverter separate from the F4 then in theory you wouldn't need to invert the signal inside the FC but I'm not sure how well those inverters handle half-duplex.
Maybe you should try the uninverted signal which has an exposed pad on recent R-XSR modules?

From https://oscarliang.com/f1-f3-f4-flight-controller/ 

> SmartPort is a half-duplex protocol, meaning the S.Port wire is bi-directional that data is sent and received in the same wire (though not at the same time, that’s why it’s only “half”).

>F3 and F7 STM MCU can handle half-duplex protocol internally in the chip itself, so you can connect SmartPort directly to these flight controllers without any modification. But F4 doesn’t have this capability.

>Although SmartPort is also inverted, F3 and F7 can invert the signal coming in or going out internally, so no problem there.

>F4 does have the half-duplex capability too, but it doesn’t work with inverted signal without an external circuit that does inversion for it bidirectionally.

teckel12, 2018-07-06:
So it sounds like with the F3 you can just flash and adjust some settings due to the internal inverter.  But with the F4 you need to flash, and rewire both the receiver and the FC to be uninverted on both ends as well change the settings.

I have a XSR-M receiver on the quad I'm considering changing to F.Port (mainly because S.Port doesn't work with the XSR-M and INAV 2.0.0).  The XSR-M doesn't have an exposed uninverted pad, you need to wire it to one of the MCU pins, which I'm not going to do.

I did find this link which explains it much better: https://www.propwashed.com/frsky-fport/

Specifically, towards the bottom where it spells out the CLI commands for each FC type.  While the Betaflight F.Port instructions talk about inversions hacks which I have none of, the instructions above are worded much better.  Since you have an F3, you use the first set, as my F4's inverter isn't bi-directional, I'm left with the final set of commands under "Any Flight Controller wired to “Uninverted” SmartPort"

I'll try connecting to the uninverted UART1 using this set of CLI commands.

Thanks!

teckel12, 2018-07-06:
Tried it again, doesn't work on un-inverted UART (doesn't appear to work unless I also un-invert my receiver).  Also tried for fun to wire it to soft serial but that also didn't work.  I'm giving up on trying to get F.Port to work on the XSR-M unless I get an F7 FC with internal inverters.

stevenverver, 2018-10-26:
Hi SPlisken Ivm having exactly the same issue. What do you mean exactly with "R-XSR was in PPM mode"? How can you switch this mode to off on the R-XSR?

putz612, 2020-03-21:
Adding this to the ticket as I was running into this as well.  Press and hold the bind button for 4 seconds AFTER its powered on.  
https://www.youtube.com/watch?v=pnihDkqifYM