iNavFlight/inav #3113: inav 1.8.1  Stats after flight incorrectly displays the max distance from home point.

AKA00TG, 2018-04-24:
Hardware: Omnibusf4 Pro
Version : Inav 1.8.1
Issue: Flight completion Stats (when disarming after landing) incorrectly displays the 'max distance' from home point. I've seen two replay videos and confirm the total distance to home point is (incorrectly formatted perhaps?) showing 1.6km when it should show 1.06, and in a second flight 1.8km when it should be 1.08
I'll assume this is a trivial issue with string formatting whereby the 'zero' is misinterpreted as white space perhaps? I know this is not critical but mention it just in case there might be a quick fix to this.

It is only a very minor annoyance. Thank You!

AKA00TG, 2018-04-24:
Is this fixed in 1.9.1?

giacomo892, 2018-04-24:
Why don't you just try?

AKA00TG, 2018-04-24:
I would but unfortunately my software development skill is too low to even consider your suggestion seriously. :(

fiam, 2018-04-26:
Fixed in #2837, which went into 1.9.0. Closing.

closed 2018-04-26 by fiam