iNavFlight/inav #3122: blheli passthrough - one of ESCs not recognized, flashing failed

mazsola76, 2018-04-25:
Hi, folks!
Same problem as in the already closed issue:
https://github.com/iNavFlight/inav/issues/1446

INAV 1.9.1
BLheli 14.9 (Flycolor Fairy 30A multirotor target)
MANA 285/295 from Gearbest (SPRACING F3 target)

After successfully flashed all ESCs from the shipped 14.3 to 14.9 due to some desync issues, the BLheli Suite recognized only ESC 1,3 & 4 - ESC 2 appeared corrupted. Reflashing was not possible after dozen  of attempts.
Switched to Betaflight all 4 ESC were identified as Blheli 14.9 and worked normally. For safety I reflashed all of them again.
Not switched back yet to INAV because it is a PITA board (BOOT jumper must be soldered, but buried deep inside the all-in-one mainboard.
Am I missing something?

Cheers,
Zsolt

Tygrys-1, 2018-05-04:
Racestar Lite 30A ESC (G_L_30), BLHeli_S 16.5  is not recognizet by BLHeli suite through INAV 1.9.1, It is reported as unknown ESC. The BLHeli suite (last) through betaflight passthrough correctly recognizes the ESC and allows programing it.

artusmuc, 2018-05-04:
i have this fiary 30A blheli 14.9  with a omnibus f4 target but no problems running oneshot and passthrough is working for me

Tygrys-1, 2018-07-16:
#3582

digitalentity, 2018-07-29:
Can't reproduce. Working on all BLHeli_S and BLHeli32 ESCs I have on my bench.

closed 2018-07-29 by digitalentity

SeanTheITGuy, 2019-07-05:
This is out of date now, but I can reproduce this with Racerstar 35A blheli_s ESCs, with a SPRacingF3 and latest  (2.2.1) iNav.  Will export config, revert to Betaflight, flash, and return to iNav.