iNavFlight/inav #3124: Current sensor not showing correct above approx. 20 amp. on OSD / Telemetry

wilco1967, 2018-04-25:
### Board and Version
INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c1922) (Tricopter)
and also on
INAV/OMNIBUSF4V3 1.9.1 Apr 21 2018 / 13:13:58 (03a5c1922) (Flying Wing)

### Behavior

Current sensor indication on OSD / Telemetry seems to work just fine at low throttle, but above say 20 Amp, is suddenly start to show much lower (like 1/10 of the actual reading)

Current sensors work just fine on 1.8.0, as well as on pre-1.9.1 development builds.

In the blackbox log, it seems to work just fine, 
But it's not correct for high readings on OSD and not on Telemetry.
Also the mAh counter, is much slower (as if the current was really low).

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/2PE39SX8 (for the tricopter).
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/1948576/LOG00002.TXT)

(yes, I know, there is some vibration problem on the ACC, causing alt-hold problems, but that's not the issue here).

shellixyz, 2018-04-26:
Please post a config dump

wilco1967, 2018-04-26:
doesn't the pastenbin work ????

anyway. here's a dump
# version
# INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c1922)

# resources

# mixer
mixer TRI

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 502 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 20
set tpa_breakpoint = 1650
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 55
set pitch_rate = 48
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

shellixyz, 2018-04-26:
Sorry I thought it was a single link to a log

shellixyz, 2018-04-26:
Looking good to me. It seems to be proportional to motor command.

wilco1967, 2018-04-26:
in the blacklog, it works fine indeed, so the actual sensor part is ok. 
However, on OSD, and on Telemetry, it drops to very low values, as soon as actual current goes high.... it seems to happen around the 20 amp marker.

wilco1967, 2018-04-26:
here's two pictures, taken 1 second apart... notice the current, versus the throttle position

first picture, 60% throttle, 1.0 amp, (believe me, it's more like 25 amps).... 
second picture, 1 second later, i've reduced to 55% throttle, showing 20.1 amp, which is probably correcy

[https://photos.app.goo.gl/W0cRwaADLomT9dwX2](url)

shellixyz, 2018-04-26:
I know it doesn't help you but it displays current > 20A fine for me. Strange indeed I don't know what the difference is between our setups.

[image]

wilco1967, 2018-04-26:
happens on two different boards so far.... one tricopter, and one plane....
Both using an external power sensor, and both Omnibus (although different versions... one Omnibus F3 (standard omnibus), and one F4V3 , the one without build-in current sensor)

I have another plane, using a Matek F722,... that works just fine (just tested up to 30 amp,  no problem) 
Amp sensor is from a Matek FCHUB-VTX

Could set current_meter_scale = 580 have something to do with it ?.... 
the Matek uses 179.... the tricopter 500, the FX61 500. The tricopter and the FX61 both have the problem.
but then again, I use 580 also on iNav 1.8.1, and it works fine, and so does the blackbox log....

shellixyz, 2018-04-26:
My screenshot comes from a craft with an Omnibus. `current_meter_scale = 580` looks fine to me. The max measured current with this value should be 56A. I checked and the way the current is displayed on the OSD between 1.8 and 1.9.1 has not changed.

ondrascz, 2018-04-26:
I confirm simmilar. FC F405-CTR. INAV/MATEKF405 1.9.1 Apr 21 2018 / 13:25:29 (03a5c1922). The max current value I can get is 55A. When the current is bigger, nonsense low value is displayed (like 1.2A etc.). It affects also the used capacity which is increased very slowly. Checked back witch 1.9.0 ( # INAV/MATEKF405 1.9.0 Mar  6 2018 / 18:21:38 () ) on the same craft and I was able to get i.e. 75A.

ondrascz, 2018-04-27:
Aditional details:
[image]

shellixyz, 2018-04-27:
I think I found the issue. Could you please test this build: [inav_1.9.1_current_fix.zip](https://github.com/iNavFlight/inav/files/1955341/inav_1.9.1_current_fix.zip)

shellixyz, 2018-04-27:
Great news ! Yes I believe it should work with your FX61 too.

digitalentity, 2018-04-27:
Fixed.

PR #3131 (2018-04-27): Fix overflow in currentSensorToCentiamps()
Fixes #3124
closed 2018-04-27 by digitalentity