iNavFlight/inav #3127: Why msp_baudrateIndex isn't always 115200?

pmarkiewicz, 2018-04-26:
On two targets msp_baudrateIndex is set to 57600, why is this?

DzikuVx, 2018-04-27:
Good question that probably will never get a answer...

fiam, 2018-04-27:
@pmarkiewicz I think the designers of those boards intend those UARTs to be used for bluetooth bridges, that's why they set those values by default. But you'd need to ask them for a definitive answer :-)

DzikuVx, 2018-04-27:
@fiam personally I kind of doubt.

fiam, 2018-04-27:
@DzikuVx I'm positive about the F35, this is from their manual:

[image]

As for the other one, MATEKF405SE, I guess we could add Sampson.