iNavFlight/inav #3135: OSD issue

Raptorframes, 2018-04-29:
_Omnimbus f4 pro
_1.9.1

_omnimbusf4pro 1.9.1

### Behavior
* _ no matter what I Navi release used, the osd has issues such as numbers above 1000 are not displayed, some of the units are not displaye, as well as the boards current sensor and v bat are not working on the osd as well as in the inav platform. When I flash betaflight all of these features work flawlessly. Shown on this video 

https://youtu.be/XuYFwazWPOo

shellixyz, 2018-04-29:
Did you update the fonts ? Font manager in the OSD tab.

mrcottonmouth, 2018-04-29:
Try this in the CLI:
set display_force_sw_blink = ON

Raptorframes, 2018-04-29:
I updated the font and that fixed my osd issue. But I am still not getting v bat, current sensor or amperage readings. I have my battery hooked up right to the board per the diagram. These work on betaflight but not inav

shellixyz, 2018-04-30:
You should see it. Maybe your screen is cropping the image. It should be on the first line. Try to lower it a bit. One or 2 lines.

Raptorframes, 2018-04-30:
But its not even reading in the top bar on inav. It works on betaflight so im very confused to say the least!

shellixyz, 2018-05-01:
Strange. I'm using an OMNIBUSF4PRO in one of my aircraft and it is working fine: voltage and current. Current and voltage are always at 0 ?

Raptorframes, 2018-05-01:
yeah, it works on beta flight so its not a hardware issue. must be in the inav software

DzikuVx, 2018-05-02:
I can confirm that OSD is working flawlessly in latest INAV. I would retry uploading fonts

nmaggioni, 2018-05-19:
@DzikuVx Did you experience blinking problems with the Matek F405-Wing? Mine needs `display_force_sw_blink` enabled, otherwise values disappear instead of blinking.

It was a known problem on my past Omnibus F4 clone, but I expected this new board to work fine with the HW setting.

closed 2018-06-06 by DzikuVx