iNavFlight/inav #3136: flashing inav using eclipse

amitmanish, 2018-04-29:
I am learning inav codebase and wanted quick modification and flashing capability. I can build inav using eclipse but could not find any guide regarding flashing using eclipse. Is it possible to add plugins to eclipse for flashing inav. I have naze32 board and i was wondering if its possible.

DzikuVx, 2018-05-01:
I'm affraid noone is using Eclipse over here. AFAIK most devs are using VS Code

bansawelite, 2018-05-14:
@amitmanish you can both build and debug in Eclipse. I have personally used the Segger J-link debugger with eclipse for live debugging. This however requires that the SWDIO, SWCLK +5v and GND pins are available on the board (You may need to solder wires to get to these pins). 
On the reverse/back of the Naze32rev 6 board there is a 10 pin cortex debug connector. This would allow debugging too.

[image]
[image]

The Cleanflight team have an article on debugging [here](https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md).
It's a its a bit outdated but still useful. If you can get the build inav or Cleanflight/betaflight in Eclipse you've already done 95% of the hard work! The last 5% should be easy for you to setup hardware debugging( I know I had issues with the ARM and build tool path spaces in windows) Had to install these tools without any spaces.

stronnag, 2018-07-22:
example using st32flash or dfu-util as appropriate https://github.com/stronnag/mwptools/blob/master/samples/flash.sh Adapt as needed.

closed 2018-10-04 by stale[bot]