iNavFlight/inav #3137: Board alligement issue

sniperxnl, 2018-04-29:
Hai guys

I have a F35 and i have him 270 degree on yaw.
When i do the 6 step calibration and i am on step3, it makes step6 light up, how is this possible, i have changed the yaw alligment

[image]

artusmuc, 2018-04-29:
Yes - thats the same for me - you have to calibrate with "original" direction of the FlightController ..

artusmuc, 2018-04-29:
If your are calibrating a Chinese Clone SPRACING F4 clone without a fight direction arrow - you have to guess some minutes how to calibrate :)

sniperxnl, 2018-04-29:
@artusmuc

This is not a clone, and there is a mark what the front side it

artempolikarpov, 2018-04-29:
> Nothing weird here. Board alignment has nothing to do with accelerator calibration. You just need to calibrate towards sky and towards ground then do vertical calibrations in any order you want.

But in 1.9.0 calibration was totally ok, whatever direction of FC was

artusmuc, 2018-04-30:
This is exactly what i what i want to say stronnag - you have to calibrate as the FlightController is marked - and snipernxl - this was an example that i have the same wrong thinking - independent which board this is. I only want to say that this kind of calbration can be a little difficult when you have mount your FC not in "natural" geometry for example have mounted your FC with -45° 
Stronnag is the it important for calibration to follow the steps 1 to 6 - or can you jump from 2 to 5 and then 3 and 4 ?

kamik2, 2018-05-05:
might this information should be visible in the configurator - calibration tab.

rcabuser, 2018-05-11:
I Have the same "problem", and i cannot understand how this will be correct when the position you hold the plane in does not correspond with the results in Inav?????

digitalentity, 2018-05-12:
@rcabuser et al please read the manual https://github.com/iNavFlight/inav/wiki/Sensor-calibration

> You may find this easier to do more accurately prior to installing the flight controller in your model and this procedure MUST be done referenced to the marked orientation on the board.

@kamik2 please open an issue for this suggection in Configurator issue tracker.

rcabuser, 2018-05-12:
@digitalentity the model i have it installed in is so small that it's no problem to do the calibration. But i guess i'm just thickheaded when it comes to understanding this, that i should calibrate by the way the direction arrow on the board is pointing??
Inav ask you in point 3 to hold the model on it's side with the right wing down, my board arrow is of course pointing down like i have the plane nose down. If i fly the plane and banking hard to the right and it goes into failsafe, wouldn't the Inav think i'm nosed down then and do corrections according to that??
@sniperxnl have you flown that thing yet?

DzikuVx, 2018-07-03:
Calibration steps order does not matter besides first one. All the other can be done in any order.

closed 2018-07-03 by DzikuVx