iNavFlight/inav #3139: Matek F722-STD Barro not working with Mag

leonvuksic, 2018-04-29:
### Please consider using other user support 
### Board and Version
Matek F722-STD iNav 1.9.1

### Behavior
* when external magnetometer is connected the barro stops working, when I remove magnetometer the barro starts working again

The mag is on a same board with ublox m8n, tested on spracingf3 working fine.

Mag is HMC5883

MATEKSYS, 2018-04-30:
There is a fake version of the HMC5883L which is actually the QMC5883.  try to select QMC5883 in configurator

leonvuksic, 2018-04-30:
Thank you Matek, I will try today and report back. I am using the build that is posted on your website with "Add Servo definition on S6/S7 pad for Quad X." 
I have one camera servo on S6 pad and another one on pwm pad.
Can I get them to work?

leonvuksic, 2018-05-01:
Can anyone help please?

leonvuksic, 2018-05-02:
Hi Matek, the wiring is good, if I swap the wires, It doesn't work, and when it works I am getting a good read.. I don't have another gps module with compass :( looks like I'm gonna need to buy another one if I want this FC to fly... I will try Betaflight, maybe both work there...

closed 2018-05-02 by leonvuksic

DzikuVx, 2018-05-02:
I also can confirm F722 working with HMC5883l. No problems of any kind so I really suspect hardware issues with that magnetometer

imrj, 2018-05-02:
am using with a MAG3110 and that also works well....but now suddenly i have a BAROMETER FAILURE message on OSD and it wont arm   :(   any ideas? the baro looks fine to me

leonvuksic, 2018-05-03:
Get rid of this FC because its a POS... going back to F3 FC everything works like a charm.. I will test with a F4 CTR tonite and post back if it works.. My guess is there is an adress conflict somwhere which is killling the barro..

john4242, 2018-05-16:
I know it's not F722, but Matek F405, but I had similar issues.

The GPS unit is APM2.6 2.8 GPS Module High Precision Ublox NEO-M8N GPS With Electronic Compass PIXHAWK from Banggood.

The GPS part works perfect! So that's good.
However, the magnetometer is strange. I did solder two pull-up 1K resistors between 3.3V (in the GPS) and SDA/SCL.
The effect is:
- I can EITHER have the magnetometer working (if it's plugged in to SCL/SDA)
- OR I can have the on-board barometer working
But never both.

I think what's happening is that the shared I2C bus is related to this issue: the on-board barometer and the external compass share the same I2C bus.
Now, the external GPS/compass has a 3.3V regulator for the compass and the pull-up is against that, maybe that's also not good.

I ordered a separate I2C compass module, and will do some tests.

leonvuksic, 2018-05-16:
Hey John,

I have Matek F722-STD, F405-CTR and F405-AIO

I alos have the exact same model of GPS/Compass you are talking about

I am also having the same issue on all 3 flight controllers,

I made a video for Banggood and I hope they will give me back store credit so I can get the Benitian BN-880

Altough, the module selling on Banggood is also a fake Ublox clone, but it has enough memory so you can flash the latest ublox firmware and it wont forget settings after poweroff like other fake modules do...

 

cheers

 

Sent: Wednesday, May 16, 2018 at 9:12 PM

john4242, 2018-05-17:
Thanks for the tip, I think I'll get that Benitian.

Meanwhile, I got my GY-271 compass module from ebay (QMC5883L).
And guess what: it works! I've connected it to the 3.3v on the F405 board.
I think the fake M8N's compass was bad/fake, and conflicted with the on-board barometer.

-----
System Uptime: 23 seconds
Current Time: 2018-05-17T20:06:17.621+00:00
Voltage: 11.28V (3S battery - OK)
CPU Clock=168MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=QMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x0, 7865344kB, 11/2017, v0.0, 'APPSD'
Filesystem: Fatal - no FAT MBR partitions
Stack size: 6144, Stack address: 0x10010000, Heap available: 1024
I2C Errors: 0, config size: 3516, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 1005, PID rate: 995, RX rate: 49, System rate: 9
Arming disabled flags: COMPASS RX CLI

leonvuksic, 2018-05-18:
Nice! Also forget what I said abou the Benitian, today I just read on Facebook that its crap.. so.. good luck!

Ravid824, 2019-05-25:
Hi Forum, FYI I found simpler solution in case that you are facing the same problem.
I faced the same problem with Matek controller F722-SE. I am using ublox m8n with HMC5883 Mag which was working for me on the spracingf3 but not on the Matek controller..
I found out that the SCL/SDA lines has 2.2K pullup resistors. I added in parallel 1K resistors (one per line) to the 3.3V and this resolved the problem.
What happens is that the wire added from the controller to the GPS (in 400KHz clock frequency set by default by the iNAV) cause bad signals due to the length which adds a lot of errors. After I added the resistors the signals looks much cleaner and it looks to be robust.
Just a tip for you.
PS: Before applying the solution **Please verify** first the existing pullup resistors. If you cant find the resistors you can measure with multimeter the resistance in OHMs.
Measure between the 3.3V (not the 5V of the main GPS connection) to the SCL and to the SCL.
The GPS board has a 3.3V regulator with supply the power to the HMC5883.
### **NOTE: If your MAG has 1K resistors you don't want to add additional pullup resistors, you have another problem.**
### **Double check your self and make sure that you are not shorting between the SCL and the SDA which might damage your controller.**

grumpyoldgit, 2020-02-11:
I'm having the same problem described here, with an f722-wing board. I've tried two different Matek M8Q-5883 combined gps-magnetometer modules, both GPSs work over uart but I get no recognition at all over i2c. I'll try the GY-271 and see if that works, but otherwise I have to suspect the board.

sdfsdf107, 2020-06-30:
I have the same problem. Will matek give some solution on his product. His gps does not suit for his fc.

grumpyoldgit, 2020-06-30:
My problem was stray capacitance between the i2c data and clock lines. I suspected it when the system was physically unstable (separate the lead wires and it worked) then confirmed with oscilloscope. I fixed it using coaxial lead wires for both lines with the sheith of the coax connected to i2c signal ground.

sdfsdf107, 2020-07-01:
I am not fomiliar with hardware . could you please share some pictures and bom you used ?

sdfsdf107, 2020-07-01:
[image]

Nevola36, 2021-09-26:
Hello everyone, in my case when I set it to 400Mhk, the magnetometer keeps going. I set it to 200Mhz and it solved. Will I have problems because of this? Would it be better to buy new GPS/Mag? 

Thanks