iNavFlight/inav #3140: MATEK F405-STD: Enable Servo Output on S7 PAD

Technikfreak2002, 2018-04-30:
Hello!

I want to setup a Servo Gimbal on Matek F405 STD board.

The pitch servo on s6 works. But i can't figure out how to setup the roll servo.
I thougt it must be s7 pad. 
I also can't resource anything through CLI.

In Betaflight 3.3 everything works fine.

Am I doing something wrong?

Thank You!

shellixyz, 2018-04-30:
What type of aircraft ? The camera gimbal should be connected to the first two servo outputs 1->pitch, 2 -> roll.

Resource remapping is not available.

shellixyz, 2018-04-30:
The MATEKF405 target only offers one servo output by default so unless you build a custom firmware you will only get an output for the pitch on S1 for the MATEKF405 target and S6 for the MATEKF405_SERVOS6 target.

Technikfreak2002, 2018-04-30:
Ok. Thank you for your fast answer. In this case i will go back to Betaflight. Maybe alt hold and poshold work there...

Technikfreak2002, 2018-05-01:
Hello!

Can you explain me how to compile a custom firmware?
Is this compiled with eclipse and gnu arm toolchain? (I found a setup guide for STM32F4 using this)

In Beataflight resourcing the servos to s5, s6 and led_strip to s7 works. but alt_hold and pos_hold is really bad.

Thank you!

alexeystn, 2018-05-01:
Actually you don't even need Eclipse or any IDE to build a custom firmware, you can compile it with a command line. All you need to install is described here: 
https://github.com/iNavFlight/inav/tree/master/docs/development

shellixyz, 2018-05-01:
@Technikfreak2002 Once you installed what's needed following the documentation @alexeystn posted and you can build the firmware you need to edit the file src/main/target/MATEKF405/target.c. Replace TIM_USE_MC_MOTOR or TIM_USE_MC_CHNFW with TIM_USE_MC_SERVO for the output you want to transform from a motor or forwarding output to a servo output. The only rule you need to follow is to never assign a to a servo output an output that uses the same timer as a motor output. For example S2 (motor output) uses timer 8 (TIM8 in first column) so you can't change S3 which uses also TIM8 to a servo output. But you can change S7 to a servo output since it is the only output using timer 4 like that:

Change:
```
{ TIM4, IO_TAG(PB8),    TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_MC_CHNFW |                    TIM_USE_FW_MOTOR },  //S7 DMA1_ST7 MT2
```
to
```
{ TIM4, IO_TAG(PB8),    TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_MC_SERVO |                    TIM_USE_FW_MOTOR },  //S7 DMA1_ST7 MT2
```

You will get 2 servos outputs. For example with MATEKF405_SERVOS6 one servo output on S6 and another one on S7. So you will get for the servo gimbal feature pitch on S6 and roll on S7.

Technikfreak2002, 2018-05-01:
Thank you very much! I will try that!
I've set up everything using WSL on Windows 10. It compiles! (without changes yet)
So lets try...

shellixyz, 2018-05-01:
Great :)

Unfortunately I don't know enough about DMAs yet to help you changing the LED_STRIP pin.

Technikfreak2002, 2018-05-03:
@shellixyz 
Thank you again for your help!

I could not get the led_strip to work on S7. So I put the servo back there. The little change in the code, you posted, did it!

Now everything works! :D

Thanks and happy flying!

closed 2018-05-03 by Technikfreak2002

haka002, 2018-10-31:
@Technikfreak2002 Can you help me with the 2 servos?
I modified the FW some times, I could set the servos to s5,6,7 but never worked 2 servos in the same time, and the S7 at all :S
So I have Matekf405-std x quad with SERVOS6 FW LED is on S5.
Could I have 2 servos in inav with this FC?

Thanks!

Technikfreak2002, 2018-10-31:
@haka002 
As far as I remember I used the SERVOS6 version with the same board. 
I made shellixyz's changes and compiled the firmware. And it worked.
But INAV at all doesn't work well so I changed back to ardupilot.

massigrohmann, 2019-05-05:
I have the same problem. Can you send me a customize firmware for matek f405std? I need only 2 servo on s5 e s6 and if you can let me to use them it is good for me.

gsixx8, 2019-05-20:
I need that custom firmware for the Matek F405-STD as well.  I am wanting to put a pan/tilt gimbal on my iNav long range multirotor.

robohippy, 2019-08-16:
iNav S5 is led output. so s6 to servo 1, s7 to servo 2 ?

Aleksandr1a1ko, 2020-02-03:
Hi, I would be grateful if you send me the firmware for 405-STD servo 6, with a working servo on s7.

SeanTheITGuy, 2020-04-16:
Is it at ALL possible to have all 7 outputs enabled?  4 for motors, 3 for servos?  I've gotten 4 and 2 to work, but there seems to be some sort of DMA conflict keeping S7 from working when 5 and 6 are enabled.