iNavFlight/inav #3142: Baro MS5611

artusmuc, 2018-05-01:
### Board and Version
HGLRC F4 V5
INAV 1.9.1
MS5611
### Behavior
* _Describe the problem_
Jumping Altitude some meters 
* _Steps needed to reproduce the problem_
Take a cold board and go out and fly - then displayed altitude will raise up some meters during flight
* _Expected Results_
No raising up displayed altitude in OSD and telemetry  while moving at the same height 
* _Actual Results_
I have now a "workaround" .. i am worming up my board by connecting to power for some minutes - then i go out and flying - the altitude works correct.

I have used this kind of baro - cause its described as "Temperatur-compensated" Baro. Seems the compensation does not really work. 

I will have in some weeks a quad  with a FC  build in BMP280 - and on my desk i have no effect of climbing altitude while the FC warm up.

DzikuVx, 2018-05-01:
I'm affraid we can not do anything about it. Protect baro with a foam / sponge

artusmuc, 2018-05-01:
Yes - i know this with foam - this is only against light and airpressure by props. Helps not in case of raising temperature. If there is no solution - no problem - i have a workaround with "worming up" the FC.

DzikuVx, 2018-05-09:
Maybe you should consider connected an external baro over I2C and to protect it is some kind of enclosure.

closed 2018-05-09 by DzikuVx

artusmuc, 2018-05-09:
Maybe sometimes happens somebody  quick close an issue - important its closed. This kind of baro is external and connected to I2C. My questions was - it has a temperature compensation. Your first answer the right for me "We can't do anything against" - so i have to live with this hint.  
Thank you

digitalentity, 2018-05-09:
Temperature compensation code for MS5611 is in place and done according to datasheet. If it doesn't do job as good as we expect - it's TE Connectivity (manufacturer of the chip) to blame.

DzikuVx, 2018-05-09:
@artusmuc I do not know what you expect us to do exactly here.

artusmuc, 2018-05-09:
Sorry for sending twice. If its not possible to do something against - its ok - i don't expect anything. 
For my decision with one is better "BPM280" - or "MS5611" i decide to buy this one with temperatur compensation. I have already seen the climbing effect in another project i have made last year with cleanflight. I have expected with INAV a better result. But i am no expert - i thought the compensation comes over dataline and can be calculated and compensated by INAV. Seems i am wrong - i have now here some external BMP280 maybe they are working better without compensation. 
Thank you for your kind answer - and all your efforts you have done for this great project.