iNavFlight/inav #3143: Feature request: Mid Flight Disarm

Dr-Drone, 2018-05-01:
I was thinking if it was possible to not have the home point reset when disarmed and have the home point saved only when powered on. This may help save the aircraft if at all it decides or we accidentally disarm in flight.

Another feature would be to have iNAV return to home maintaining the current altitude and once it reaches home it could then lower it's altitude.

stronnag, 2018-05-01:
I'm sure we discussed this before (can't recall where) and came to the conclusion that the best solution would a setting along the lines of:
```
set  inav_reset_home =  FIRST_ARM
Allowed values: NEVER, FIRST_ARM, EACH_ARM
# just like inav_reset_altitude
```
Or maybe one `inav_reset_important_stuff` setting that does both.

Dr-Drone, 2018-05-02:
Yeah that seems just about right.

danarrib, 2018-05-07:
This is a VERY important feature. Anyone's planning to do it?

OlivierC-FR, 2018-05-07:
I second this suggestion !
Mid-air disarms are annoying enoughnalready, but then you also lose the direction and distance to home, and that's not helping at all.

Dr-Drone, 2018-05-08:
Nice! 👏

fiam, 2018-06-07:
Fixed in #3143

PR #3183 (2018-05-09): Add new parameter for controlling home position reset
Add new inav_reset_home setting with the same possible values and
semantics as inav_reset_altitude (as suggested by @stronagg),
but with a default value of EACH_ARM to keep backwards compatibility.

Fixes #3143
commit 954ae180 (2018-05-09): Add new parameter for controlling home position reset

Add new inav_reset_home setting with the same possible values and
semantics as inav_reset_altitude (as suggested by @stronagg),
but with a default value of EACH_ARM to keep backwards compatibility.

Fixes #3143
closed 2018-06-07 by fiam