iNavFlight/inav #3144: Current sensor on Matek FCHUB-W appears to be overflowing/rolling over on iNav 1.9.1?

StewLG, 2018-05-01:
I'm setting up a Matek F405-STD + FCHUB-W pdb combination on a Mini Drak running iNav 1.91. This is a fixed wing PDB:

http://www.mateksys.com/?portfolio=fchub-w

The Matek directions specify a Scale of 317 for iNav (I'm assuming), and that gives me reasonably linear, reasonably accurate amperage up to about 33 amps as compared to a bench watt meter. Above that, the indicated amp value rolls over to 0, going up from there to maybe 12-15 (unsure, don't like to push this motor for too long). The motor draws about 55 amps at full throttle according to the bench meter.

The same number is displayed in the iNav configurator as in the iNav OSD so I'm assuming this is not just a display issue in the OSD, for example.

If I put the scale back to the shipping default of 179, I can get a higher amp number out of iNav, although it will be inaccurate. So at the same throttle position I roll over at ~33 amps, at 179 I get about 58 amps before it rolls over and continues up from there.

Matek indicates their current sensor is good to 104A.

My guess would be iNav is expecting only 3.3 volts on the current sensor but the FCHUB-W goes to 5 volts. But that's just a guess, and I haven't looked at the code yet.

Suggestions for how to debug this in a minimally-invasive fashion would be appreciated. I can attempt debugging this myself, but unfortunately the F405-STD seems to lack SWD debug pins. (Maybe they are offered on the ribbon cable? Matek can you help?)

I did search on the repo for similar issues but did not find any. If I've missed something please point me to it and accept my apologies.

Thanks!

shellixyz, 2018-05-01:
Duplicate of #3124. It is a bug that has already been fixed. You can use this build while waiting for version 2.0: [inav_1.9.1_current_fix.zip](https://github.com/iNavFlight/inav/files/1955341/inav_1.9.1_current_fix.zip)

To debug you don't always need the debug port. You can use the DEBUG_TRACE facilities (see https://github.com/iNavFlight/inav/blob/master/src/main/build/debug.c) or the `debug` array. The DEBUG_TRACE messages can be displayed by clicking on "Open debug trace" in the configurator's sensors tab. The values of the debug array can be displayed in the configurator's sensors tab by checking the debug checkbox.

StewLG, 2018-05-02:
Ach, I only searched open issues when I looked for this. The patched binary works fine for me - thank you for that, and for the console debugging tips.

Again, my apologies for the duplicate issue.

closed 2018-05-02 by StewLG