iNavFlight/inav #3155: Accelerometer Calibrationproblem on F3 10DOF

jonastogo, 2018-05-03:
I installed iNav on my F3 Deluxe yesterday and wanted to calibrate the accelerometer.
iNav is on version 1.9.1.
If I now go through the 6-point method according to the instructions, the calibration does not work.
In step 1 nothing happens after calibration, which means there is no check mark in step 1.
If I do step 2, step 5 is completed.
For all other steps, it is a matter of luck whether anything at all is set to complete.
But if it's never the right thing.
No settings have been changed, i.e. the software is set to factory settings.
Calibration worked perfectly with Cleanflight.
With version 1.9.0 I have the same problem.

Does anyone have an idea and can you help me?

artusmuc, 2018-05-03:
Yes - since 1.9 you see the result of the calibration and you have to calibration like the Flightcontroller is "natural" marked. There is a normally a white arrow on every FC. This is your natural calibration and this result is shown in calibration tab.

jonastogo, 2018-05-03:
I did that.

jonastogo, 2018-05-03:
Okay, I'll try to describe what I did.
For step 1 I laid the Fightcontroller flat on the ground.
The arrow was on the top and pointed away from me.
Then I clicked "Calibrate Accelerometer".
After the calibration was completed, step 1 was not marked as completed.
For step 2 I put the flightcontroller flat on the ground, but with the arrow at the bottom. The arrow continues to point away from me.
After calibration, step 5 was marked as connected and not step 2.
For the remaining steps I put the FC on the edge. The underside then pointed to me.
Nothing happened at any step but step five.
In step 5, step 6 was completed.

I really don't know what I'm doing wrong.

jonastogo, 2018-05-03:
In the sensor tab, I see the values from all the sensors going crazy.
I compared the values of the sensors on iNav and Cleanflight at the same position of the controller.
iNav Accelerometer:
X: -0.17 , Y: -1.08 , Z: 0.80
Cleanflight Accelerometer:
X: 0.00 , Y: 0.00 , Z: 0.80

iNav Magnetometer:
X: 0.39 , Y:-0.05 , Z: 0.44
Cleanflight Magnetometer
X: 0.40 , Y:0.00 , Z: 0.00

iNav Barometer
Alt: ~-6.5 , Baro: -0.3 - -0.8
Cleanflight Barometer:
Alt: ~-0.4

The sensor data of Cleanflight look relatively precise to me, in contrast to iNav.
So, it looks like iNav can't read the sensors properly.

jonastogo, 2018-05-03:
No Problem :)
Yes i did a full erase. Is that wrong?

jonastogo, 2018-05-03:
I've tried multiple version of inav and inav configurator now.
It is not working at all.
The Accelerometer can not be calibrated.
When the Controller is flat, the simulated copter is in an angled position.
The barometer give me crazy results aswell.
Sometimes the Altitude is increasing or decreasing slowly when i dont move the flightcontroller (and no wind), sometimes the altitude value stays at one level.

With Cleanflight everything is working fine, but there is obviously no GPS.

artusmuc, 2018-05-03:
do you try betaflight ?
no full erase solve sometimes problems:)

jonastogo, 2018-05-03:
Betaflight is working fine aswell :)
no full erase does not work.

digitalentity, 2018-05-04:
What board is this exactly and which firmware are you using?

Make sure you use latest Configurator available here: https://github.com/iNavFlight/inav-configurator/releases/tag/1.9.3

First step needs to be in the position that lights up the first checkmark. Betaflight/Cleanflight can't be used as a reference since they don't really calibrate the sensor chip itself.

jonastogo, 2018-05-04:
I'm using a SP Racing F3 Deluxe
MPU6050 accelerometer/gyro
MS5611 barometer (Deluxe only)
HMC5883 compass (Deluxe only)

I'm using the Configurator Version 1.9.3 with iNav 1.9.1

So when i hold the Controller to check the UAV test, the controller is heavily angled.
When i calibrate, should i hold the controller heavily angled aswell?

digitalentity, 2018-05-05:
Please place the FC flat, level and not moving and check the values in Sensors tab (specifically Accelerometer). Post a screenshot here.

digitalentity, 2018-05-05:
You have a faulty accelerometer sensor. Uncalibrated sensor should read roughly zero on X and Y and approximately 1G on Z-axis. In your case a flat FC is reading huge acceleration on Y axis. This means there's likely internal damage to the micro mechanics of the sensor. It's simply unsafe to fly this aircraft.

jonastogo, 2018-05-05:
In Cleanflight all data is fine without calibrating.
This is confusing to me.
Do you think its possible to cut the chip out and implement a new one?

digitalentity, 2018-05-05:
Cleanflight doesn't care about sensor data sanity, symmetry and scale, it just records sensor offsets when you press "Calibrate" once. This "calibration" is only valid for one position - being perfectly level, that's it. In all other positions this "calibration" will yield results that don't correlate with reality at all.

Sensor can be replaced if you are good with hot-air soldering machine, but I'd just buy a new (better, preferably F4-based) board.

S-T-A, 2018-05-08:
I am having problems with the same firmware and configuration but with a colibri race f3 first calibration worked updated firmware and it won't get past first calibration???

jonastogo, 2018-05-08:
I've just got a new SP Racing F3 Deluxe and it works fine now :)

FrozenDroid, 2018-05-11:
Can this be closed?

digitalentity, 2018-05-12:
Confirmed to be  a hardware issue.

closed 2018-05-12 by digitalentity