iNavFlight/inav #3157: Pwm 5 for servo cant use

abdul1212, 2018-05-04:
Hello my friends, i use airbot f4

Pwm 1-4 is active i use wing in inav.
In pwm 5 i wont use servo for my camera fpv. So i can turn around. But pwm 5 not active.

Uart 1 i use gps
Uart 6 for sbus

Help me please.

abdul1212, 2018-05-04:
[image]

abdul1212, 2018-05-04:
Thats my fc, airbot f4

DzikuVx, 2018-05-13:
CLI dump please

abdul1212, 2018-05-15:
Entering CLI Mode, type 'exit' to return, or 'help'

# DUMP

# version
# INAV/OMNIBUSF4 1.9.0 Mar  5 2018 / 00:12:35 (dc4ef594f)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -125 -1 
servo 4 1000 2000 1500 125 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 494 lines cut ...]
set nav_mc_vel_xy_p = 40
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 33
set tpa_breakpoint = 1300
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

closed 2018-07-29 by stale[bot]