iNavFlight/inav #3158: mmix issue on yawing

vc, 2018-05-04:
### Board and Version
 INAV/OMNIBUSF4V3 1.9.1 Apr 21 2018 / 13:13:58 (03a5c1922)

### Behavior
* _Describe the problem_
I use the classic quadcopter on the X scheme. the only difference is the additional two motors that I want to use for yawing.
I created a custom mixer for an additional two motors on yaw.
motor 4 and 5 are used for yawing. with the other motors (0-3), I removed the values by yaw to zero.
the problem is that when I request roll or pitch, then PWM appears on the 4th and 5th motors.

* _Steps needed to reproduce the problem_
\# mixer
mixer CUSTOM
mmix reset
mmix 0  1.000 -1.000  1.000  0.000
mmix 1  1.000 -1.000 -1.000  0.000
mmix 2  1.000  1.000  1.000  0.000
mmix 3  1.000  1.000 -1.000  0.000
mmix 4  0.0001  0.000  0.000 -0.500
mmix 5  0.0001  0.000  0.000  0.500

* _Expected Results_
at roll and pitch yaw motors should not react.

* _Actual Results_
at roll and pitch, the yaw motors react approximately 70%.

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/xpA8Nq30

DzikuVx, 2018-05-04:
It works like expected. Throttle weight is so small that they just would
not work

vc, 2018-05-04:
why? 

look at this:
[image]

and this:
[image]

what does the "Throttle weight is so small"?

digitalentity, 2018-05-05:
Mixer is allowed to raise (or lower) throttle value to allow correction requested by PID controller to be executed. 

In your example. PID controller is likely requesting correction of 436 on `ROLL` (split into +218 and -218). 

However your base throttle is at zero (1150) meaning the negative correction won't be executed and thus correction requested by PID won't be fully satistied. To execute the correction mixer raises throttle to 1368 which allows it to execute full extent of requested correction. 
```
1368 + 218 = 1586
1368 - 218 = 1150
```

Mixer behaves exactly as it should. I think we can close this issue as non-relevant.

vc, 2018-05-07:
> Since you don't have anything but throttle on motors 5 & 6 - you see only throttle value there.

But on 5 and 6 motors the throttle is absolutely insignificant. And this is only because it can not be assigned to zero. These motors have thrust in the horizontal plane. Throttle here is really zero.

How do I create mmix settings so that the mixer is working correctly?
On 5 and 6 motors I need to have only traction on request for yaw.

This is my setup:
[image]

digitalentity, 2018-05-10:
@vc I think it would be better to place yaw control motors on servo outputs and configure servos for range [1000;2000] and middle of 1000:

```
# mmix
mmix 0  1.000 -1.000  1.000  0.000
mmix 1  1.000 -1.000 -1.000  0.000
mmix 2  1.000  1.000  1.000  0.000
mmix 3  1.000  1.000 -1.000  0.000

# smix
smix 0 0 2 100 0
smix 1 1 2 -100 0
```

vc, 2018-05-10:
Thank you for the proposed solution!
I also thought about such a solution.
But here the downside is that the servo is PWM in the DISARM state.
As far as I know, this can not be disabled by the settings in the current version.

digitalentity, 2018-05-10:
Maybe a possible solution would be to add a "disarmed value" for servos? @DzikuVx @fiam @stronnag what do you think?

vc, 2018-05-21:
Maybe there is some news in this matter?

DzikuVx, 2018-05-21:
@vc have you tried `smix` solution @digitalentity proposed?

fiam, 2018-05-21:
@DzikuVx  I think the problem is that the servos (in this case connected two motors) will not be kept idle while the craft is disarmed, since we always allow servos to move.

vc, 2018-05-21:
@DzikuVx 
> @vc have you tried smix solution @digitalentity proposed?

I do not like this varant. I checked that on the betaflight firmware this behavior is not observed with the same value of mmix.
I still do not understand why at the given mixer in mmix yaw, motors 5 and 6 behave this way.
there was a thought that such behavior on them is due to the active airmode, but this mode is disabled in the configuration tab.

for me the best option would be a solution using mmix settings. maybe you can tell me how to properly configure it in my situation? Maybe I do not understand something?

closed 2018-08-04 by stale[bot]