iNavFlight/inav #3160: Impossible midair arming with mission loaded in case the starting point is far from the craft position.

Tygrys-1, 2018-05-04:
If I am far from the mission starting point the plane will not arm after loading a mission. It is a safety check and seems ok. But if I accidently disarm the plane in the air and the plane is far from the mission starting point (and the mission is loaded) it will not arm again - which is bad.

So I suggest to refuse a mission start (NAV WP mode) in case when the plane is far from the mission starting point and not refuse the craft arming.

Tygrys-1, 2018-05-14:
It might apply also to the case when the number of sats falls below the default 6. IMHO it should always allow to re-arm if actually flying and has been disarmed by the switch or sticks.

digitalentity, 2018-05-15:
Suggestion: modify `nav_extra_arming_safety` setting to have more options: `OFF`, `EACH_ARM` (default), `FIRST_ARM`. The last setting will enforce arming safety for first arming, but will allow re-arming even if checks fail.

Tygrys-1, 2018-05-15:
This is better than nothing. :-) 
But it is still good to have safety chenck on the ground, and no limits in the air when re-gainng the controll. Maybe re-arming should be enabled unconditionally up to 10 (or so) seconds after a dissarm?

ninja-zx11, 2018-12-02:
Hi all, Can we rearm  midair on Inav 2.0.1? I accidently disarmed my quad and couldn't rearm it.
I now have done set small_angle = 180.Do you guys think i should be able to rearm midair with set small_angle = 180 on Inav 2.0.1?Thanks.

shellixyz, 2018-12-02:
@ninja-zx11 With `small_angle` set to 180 you should be able to. Unless you're using a GPS and lost the fix.

ninja-zx11, 2018-12-02:
Thanks for the confirmation!!