iNavFlight/inav #3171: Crossfire Serial Bridge (Bluetooth) stops working if craft is armed

KayCox, 2018-05-06:
Got my crossfire rx connected over UART for using serial bridge from the CRSF TX to my iPad running mobile fight app. All works fine, got full telemetry and can do any changes via the app. As soon as craft is armed, the transmission closes completely. No more telemetry is received.  
 Any ideas?

# dump

# version
# INAV/OMNIBUSF4PRO 1.9.0 Mar  5 2018 / 00:15:43 (dc4ef594f)

# resources

# mixer
mixer FLYING_WING

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 2 0 -100 0
smix 1 3 1 -85 0
smix 2 4 1 85 0
smix 3 3 2 75 0
smix 4 4 2 75 0

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1538 100 -1 
servo 4 1000 2000 1596 -100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature CURRENT_METER
[... 510 lines cut ...]
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 40
set pitch_rate = 27
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

#

Tygrys-1, 2018-05-06:
Have you enabled MSP telemetry on this port?

fiam, 2018-05-06:
You have enabled MSP and something else on that port. Enable MSP and nothing else.

stronnag, 2018-05-06:
The something else is MAVLINK. It requires that whatever the GCS is to support it. I don't think Mobile Flight does.

digitalentity, 2018-05-06:
Important thing to remember about enabling MSP together with telemetry (MAVLink, LTM etc) on the same UART is that they don't work together at once.

If craft is disarmed - port is connected to MSP
If craft is armed - port is connected to telemetry and disconnected from MSP

jpbarraca, 2018-05-15:
INAV Configurator could show a warning with this information when the port has multiple purposes.

closed 2018-06-14 by DzikuVx