iNavFlight/inav #3181: Current Sensor reporting incorrect amperage

slavon4ever, 2018-05-07:
### Board and Version

SP Racing F3 Evo, current sensor chip is ACS712 30A.

INAV Configurator 1.9.3 Firmware Version 1.9.1

### Behavior

After upgrading from 1.7.2 to 1.9.1 current sensor starts reporting incorrect values.
I've  downgraded back to 1.7.2 and upgrade each firmware version by version:
1.7.2 - OK (standby 0.3A)
1.8 - OK
1.9.0 RC2 - OK
1.9.0 - OK (standby 0.3A)
1.9.1 - FAIL (standby 32A)

My current sensor settings:
```
set current_adc_channel = 2
set current_meter_scale = -667
set current_meter_offset = 2540
set current_meter_type = ADC
```

If I change current_meter_offset to 440 the sensor shows 0.3A on standby, but when I start motor it shows from 0.4 to 6A and than goes to negative amperage.

shellixyz, 2018-05-07:
More than likely a duplicate of #3144 and #3124. Please test this build: [inav_1.9.1_current_fix.zip](https://github.com/iNavFlight/inav/files/1955341/inav_1.9.1_current_fix.zip)

slavon4ever, 2018-05-09:
Thanx. Fixed firmware works well.

fiam, 2018-07-08:
Fixed by #3293, closing.

closed 2018-07-08 by fiam