iNavFlight/inav #3194: FPort telemetry with Matek F405-STD not working

lo4wai, 2018-05-10:
### Board and Version
Matek F405-STD
INAV v.1.9.1

### Behavior
I connect the uninverted smart port pad of R-XSR receiver to TX port, and configure FPort as serial rx.
The RC link work fine, but my X9D+ cannot discover telemetries other than Rxbt and RSSI.

I have tried all UART ports, RC link work but not telemetry.
I tested the UART with TRAMP vtx and it work fine.  I believe that bidirectional serial is working on these UARTs.

FPort telemetry work on my other boards such as Matek F405-OSD or Omnibus F4 v3.

Below is the diff with the uninverted smart port connected to UART 4 tx.

> # diff
> 
> # version
> # INAV/MATEKF405 1.9.1 Apr 21 2018 / 13:25:29 (03a5c1922)
> 
> # resources
> 
> # mixer
> mixer CUSTOMTRI
> mmix 0  1.000  0.000  1.333  0.000
> mmix 1  1.000 -1.000 -0.667  0.000
> mmix 2  1.000  1.000 -0.667  0.000
> 
> # servo mix
> smix 0 5 2 100 0
> 
> # servo
> servo 5 1000 2000 1500 -93 -1 
> 
> # feature
> feature GPS
> feature LED_STRIP
> feature BLACKBOX
> feature PWM_OUTPUT_ENABLE
> 
> # beeper
> beeper -ON_USB
> 
> # map
> 
> # name
> 
> # serial
> serial 0 0 115200 38400 0 115200
> serial 1 2 115200 115200 0 115200
> serial 2 0 115200 57600 0 115200
> serial 3 64 115200 57600 0 115200
> serial 4 4096 115200 57600 0 115200
> 
> # led
> 
> # color
> 
> # mode_color
> 
> # aux
> aux 0 0 0 1300 2100
> aux 1 1 1 1300 2100
> aux 2 20 0 1675 2100
> aux 3 3 2 1675 2100
> aux 4 9 2 1300 1700
> aux 5 8 2 1700 2100
> aux 6 11 4 1725 2100
> 
> # adjrange
> 
> # rxrange
> 
> # master
> set acc_hardware = MPU6500
> set acczero_x = 7
> set acczero_y = -8
> set acczero_z = -59
> set accgain_x = 4095
> set accgain_y = 4091
> set accgain_z = 4101
> set mag_hardware = HMC5883
> set mag_declination = -250
> set magzero_x = -48
> set magzero_y = -28
> set magzero_z = -31
> set baro_hardware = BMP280
> set pitot_hardware = NONE
> set rssi_channel = 16
> set serialrx_provider = FPORT
> set serialrx_halfduplex = ON
> set min_throttle = 1010
> set max_throttle = 2000
> set motor_pwm_rate = 2000
> set motor_pwm_protocol = ONESHOT125
> set failsafe_procedure = RTH
> set vbat_scale = 1093
> set vbat_max_cell_voltage = 420
> set battery_capacity = 3000
> set battery_capacity_warning = 1800
> set battery_capacity_critical = 900
> set current_meter_scale = 151
> set current_meter_offset = -3
> set servo_pwm_rate = 330
> set small_angle = 180
> set gps_sbas_mode = MSAS
> set nav_extra_arming_safety = OFF
> set telemetry_inverted = ON
> set smartport_fuel_unit = PERCENT
> set osd_main_voltage_pos = 2168
> set osd_rssi_pos = 2116
> set osd_throttle_pos = 225
> set osd_vtx_channel_pos = 88
> set osd_current_draw_pos = 2232
> set osd_mah_drawn_pos = 2264
> set osd_gps_speed_pos = 2243
> set osd_gps_sats_pos = 2147
> set osd_home_dir_pos = 2411
> set osd_home_dist_pos = 2414
> set osd_altitude_pos = 2275
> set osd_vario_pos = 2231
> set osd_main_cell_voltage_pos = 2200
> set osd_throttle_auto_thr_pos = 2370
> set osd_heading_graph_pos = 2154
> 
> # profile
> profile 1

lo4wai, 2018-05-11:
Finally got it work by using the original inverted smart port pin from the R-XSR and connected to uart4, enable softserial and set serialrx_inverted to ON.  It doesn't work if the receiver is connected through the uninverted pad.

I found that uart pads in this board is not directly connected to the processor pin, but through a resistor instead.  I guess signal from FC after passing through the resistor may not be strong enough to be recognized by the rx without the inverter.

rhwilr, 2018-05-12:
Thanks, @lo4wai this helped a lot! I just had the exact same issue.
I shorted out the resistor by soldering a thin wire over it. This solved the problem for me.

fiam, 2018-05-12:
@lo4wai Thanks for figuring this one out. Not closing it yet, since we might to mention this workaround in the wiki.

fl0zan0, 2018-05-13:
hi do you have the cli commands to do that ?

closed 2018-07-26 by stale[bot]

HJOGermany, 2018-08-12:
I have got the same Problem with the F405STD.
Can somebody tell me (for a noob) what i have to do?

HJOGermany, 2018-08-14:
Can you please make a picture of the wired FC, which resistor is It?
I have the same problem with my setup. 
Radio link is ok. Telemetry not.

The instructions of the Matek Site doesn't work.
The rxsr is flashed with the newest FPort Firmware

Thx

teckel12, 2018-08-14:
@HJOGermany Are you connecting via soft serial?

HJOGermany, 2018-08-14:
Mo, i connected it Line Show in the Matek Site

teckel12, 2018-08-15:
@HJOGermany According to this thread, that isn't possible due to a resistor added to the FC. Instead, you need to use soft serial. See the solution in previous posts.

HJOGermany, 2018-08-15:
Ok Thank sou
Problem is fixed

nateof8, 2019-10-05:
> Thanks, @lo4wai this helped a lot! I just had the exact same issue.
> I shorted out the resistor by soldering a thin wire over it. This solved the problem for me.
After weeks of frustrating troubleshooting, this fix also worked for me!  I'm hooked up to TX3 and bypassed the 100 ohm resistor between the TX3 pad and the STM32.  Thanks a ton!!!

agetommy, 2021-01-23:
To bypass the 100ohm resistor on the TX3 pin (C10) for softserial 11 want work in my environment (R-XSR on the designated S.Port connector). A swap to the TX4  pin is the soloution in my case.