iNavFlight/inav #3195: Do not use throttle when RTH triggers with thr low (plane)

ABLomas, 2018-05-11:
Previous INAV versions had feature (not a bug ;-) - if RTH is triggered (manual switch or failsafe) AND thr was low - autopilot would not use throttle, only control surfaces for navigating home. Many autopilots use the same logic and this is perfect for unplanned landings with empy battery and so on.

But in current (and 1.8.x branch) logic changed, INAV will use throttle in any case, which complicates things a lot.
So, i see two solutions for this problem:

- add CLI switch for switching autopilot logic - "current" (use throttle in any case) and "old" - "do not use throttle if autopilot was activated with throttle low"
 - add additional logic - do not use throttle on autonomous modes if battery is below some percentage - complex fix and i think it would have some unexpected side effects (but would allow emergency landings too)

P.S. discussed in telegram inav channel, examples why current logic is bad provided here, but could copy to issue, too, if required.

romanas, 2018-05-11:
if RTH triggered by switch (not LOST rc link or other factors) and throttle is below 5% (some has bad trims), go fully RTH with no motor usage. But when adding throttle go normal RTH[L] as planed and declared in min max speeds.

digitalentity, 2018-05-12:
I believe failsafe should always have motor control. Otherwise a glide with no power might end up badly if you fly out of range.

As for the manual RTH option - I think this should be configurable.

ABLomas, 2018-05-12:
@digitalentity i disagree here.
There's video on telegram channel 20:45 yesterday with valid use case, when RTH sent plane in tree. At least 4 similar incidents around in season ;-)
In short, few examples:
 - you turned back from long trip, but wind changed direction and you see that you can't fly home. You find nice place to land, without obstacles, direct plane here and prepare to drive and pick it (using VTX as beacon for additional location data). But few meters from ground RC link is lost, RTH kicks in and plane tries to fly home, battery not enough - it ends few hundred meters away from expected location, possibly in tree, with dead battery.
 - you want to do "max-range" trips - fly far away, land here, friend changes battery, launch from hill and go home. With earlier setup it's easy, with current it's impossible (unless RTH is turned off completely).
.... and so on.

Just make it configurable, not default behavior, but keep older functionality. Now this "fix" for non-existent bug is one-way battery and plane killer.

digitalentity, 2018-05-12:
@ABLomas It's not a non-existent bug. We made RTH ignore motor stop because of multiple reports of crashes when engaging RTH (or going into failsafe). So the bug is perfectly valid for some of the flyers. Apparently you want other behavior.

I suggest implementing an CLI option `nav_ignore_motor_stop` with choices `NEVER` (what @ABLomas is requesting), `FS_ONLY` (only when RC link is down), `PILOT` (only when RC link is active), `ALWAYS` (default, navigation will always disregard `MOTOR_STOP` and throttle position)

romanas, 2018-05-12:
It has to work fully as is now, but only without motor if thr is at zero. 
and needs not to disarm or do other things (like reset GPS fix home) when thr is down.

ABLomas, 2018-05-12:
Another possible option would be "disable failsafe" switch.
When enabled - would set "failsafe_procedure" to NONE, ignoring current AUX channel value. Should be easy to implement, no?
(still, i think that setting based on throttle pos. is better option)

ABLomas, 2018-05-13:
Another idea - make failsafe mode configurable!
Now ppl who test waypoints set failsafe procedure to NONE. While this is fine fo testing - it's dangerous for everyday flying.
Moving into mountains of flying in forest - better set failsafe procedure to LAND, not RTH
....and so on...

So, adding switchable failsafe procedure to switch would resolve all mentioned issues:
 - TX switch can switch available failsafe modes
 - last mode before link loss is used, not failsafe values

So, need to land with empty battery or wanna go autonomous? Switch failsafe procedure to NONE. Wanna clear that gap, but worry about link? Switch to "LAND". In all other cases - go with RTH (or smth like that).

giacomo892, 2018-05-13:
Indeed having the possibility to set FS as land while doing acro in low altitude near obstacles would be awesome. In such scenarios start climbing can mean wing trapped on a tree :P

danarrib, 2018-05-25:
I am one of the people who had problems with the old RTH/FS behavior. I did a VERY LONG walk of shame when I lost the RC Link and the plane did the RTH without motor... Wind made RTH without motor impossible and plane just landed 1 km away from last known location.
I guess the @digitalentity suggestion of a `nav_ignore_motor_stop` covers it all. It's perfect and makes everyone happy.

romanas, 2018-05-25:
im talking about controlled rth with no throttle. if you lost rc link let normal fs with motor and throttle do theyr job. but if you manualy made RTH and removed throttle let it glide! then if you add throttle let it climb back to rth activation height. everybody will be happy.

danarrib, 2018-05-26:
Ok now I guess I understood:

- If Failsafe RTH: 
Always use motor power. Always climb to rth altitude.
- If RTH without Failsafe: 
If throttle < 5%, do not power motor and don't try to climb to rth altitude.

It's actually a good idea.

romanas, 2018-05-26:
rth without motor should try to maintain height of activated rth (hot air winds).
But mainly this topic is not about thr RTH at zero. i needed to read Ablomas text more careful =)

ABLomas, 2018-05-27:
There's no need to maintain altitude if returning, for example, from 2km altitude and 2km distance. Current code will keep motor running even at this extreme angle.
We do not ask to make swichable motor modes as default. But keep it please as option, at least as it was earlier.
-1 plane today, not exactly inav fault, but precisely burned remains of battery when link was lost, making recovery next to impossible.

digitalentity, 2018-06-10:
@ABLomas would an option `nav_ignores_motorstop` defaulting to YES suffice?

ABLomas, 2018-06-10:
Yes, if this would restore "pre-1.8 inav" behavior - it would be nice.
Based on comments it should be not default.

TL;DR version: if throttle was at zero before activating RTH (by switch or failsafe) airplane should not be using throttle but still try to navigate back home (plane leveled and gliding to home direction).
Very usefull for high-altitude flights, too (3km alt, 1km to home - airplane still use throttle to reach home pos, using battery, when it could just glide ~10km easily).

digitalentity, 2018-08-09:
Fixed by #3349

closed 2018-08-09 by digitalentity