iNavFlight/inav #3198: "sensor lost" with FPort when GPS fix captured

lo4wai, 2018-05-11:
FPort telemetries are fine before GPS fix (i.e. not enough satellite signal captured)
Taranis telemetry screen show "*" in all data except those GPS related, which indicate data is received.

When GPS fix, GPS related fields get "*"s, but others are gone, and taranis reported sensor lost after some period of time.
Telemetries other than GPS related data occasionally receive updates, therefore taranis keep shouting "sensor lost"

fiam, 2018-05-12:
Reproduced on the bench. Working on a fix.

fiam, 2018-05-12:
@lo4wai seems like not sending a sensor every 3 cycles workarounds the problem. Tested with 4, 5 and 6 too, but anything > 3 still causes some sensor lost alarms on my radio when using an XSR.

lo4wai, 2018-05-14:
@fiam glad to see your work around, I will try to build it when I have time.  I have read the code before but can't figure out why the GPS_FIX state will trigger the problem.  In fact the problem also occurs occasionally even without GPS_FIX, but with a much lower frequency.

With your fix, I wonder if the problem is related to the baud rate, since FPort is running with 115200 but smart port is 57600.

lo4wai, 2018-05-17:
Tried the work around and it works!  Thanks for the fix.

kev018, 2018-05-17:
I have also the sensor lost massage with GPS fix, must i wait for INAV 2.0 or is there an update earlier?

fiam, 2018-05-17:
My hunch is that the telemetry bandwidth from the RX to the TX doesn’t have a data rate big enough to support sending one sensor packet every 9ms. Before 2.0, the GPS related sensors where only sent once the fix was acquired, so the RX could cope with that because there were some pauses in the FC->RX stream (the slot for the GPS data was reserved but not used). Once a GPS fix was acquired, we started sending a sensor on almost every 9ms cycle and the buffer in the RC eventually got full, causing some of the sensors to be discarded by the RX.

fiam, 2018-05-17:
Forgot to mention that using Smarport would send a sensor to the RX from the FC every 22ms, that’s why we didn’t have this problem until we started supporting FPort.

P-I-Engineer, 2018-05-18:
i'm trying to make my own build with this fix in it. I'm github/code rookie.

is this the process?
git clone https://github.com/iNavFlight/inav/
cd inav

go to local directory where it downloaded
navigate to telemetry
open smartport.c
add in the lines of code with a text editor
save

back to bash
make TARGET=MATEKF405

i did all that, flashed my new hex and still have sensor lost. Do i need to be in a different folder before i do the make?

thank you for the help, it's been a long week of trying to figure this stuff out.

lo4wai, 2018-05-18:
i use

git cherry-pick 3625f05
to get the code after cd to inav directory

then build as usual with docker

P-I-Engineer, 2018-05-18:
thank you for teaching me that. 
I did a cherry pick build and am still getting sensor lost alarm.

any thoughts, from anyone?

my equipment:
matekf405
x4r and r-xsr
taranis x9d (non plus)

thanks

fiam, 2018-05-18:
@P-I-Engineer I don't have any of those receivers. Try editing smartport.c and changing `SMARTPORT_REST_PERIOD` from 3 to 2.

fiam, 2018-05-18:
@lo4wai Which RX did you test with?

P-I-Engineer, 2018-05-18:
i edited my comment before, think it wasn't working because of my lack of github skills.

i just tested with a X4r, i'll try with a r-xsr later this afternoon.

lo4wai, 2018-05-20:
@fiam I'm using R-XSR
With SMARTPORT_REST_PERIOD set to 3, it still report sensor lost occasionally.
With 2, no further lost

fiam, 2018-05-20:
@lo4wai Thanks for testing. I'll make a new PR that introduces this workaround only for FPort, since we're effectively reducing the bandwidth of the sensors by 33% I'd rather not add the extra delay for SmartPort too.

P-I-Engineer, 2018-05-28:
i flew today for a 30 minutes, no sensor lost warnings

PR #3206 (2018-05-13): Make SmartPort sensors rest a cycle out of every 3
Fixes #3198
commit 3625f058 (2018-05-13): Make SmartPort sensors rest a cycle out of every 3

Fixes #3198
commit f7f8c46d (2018-06-01): Make FPort stop one sensor cycle every 3

Sending 3 cycles and then pausing once still produces some sensor
lost warnings after 10-15 mins, but reducing it to 2 sensors/1 pause
seems to get rid of the problem. Since this reduces the effective data
rate by 33%, changes have been removed from the SmartPort driver and
moved to affect just FPort. This way, S.Port doesn't have suffer any
update rate reductions.

Fixes #3198 (hopefully for real this time)
PR #3308 (2018-06-09): Make FPort stop one sensor cycle every 3
Sending 3 cycles and then pausing once still produces some sensor
lost warnings after 10-15 mins, but reducing it to 2 sensors/1 pause
seems to get rid of the problem. Since this reduces the effective data
rate by 33%, changes have been removed from the SmartPort driver and
moved to affect just FPort. This way, S.Port doesn't have suffer any
update rate reductions.

Fixes #3198 (hopefully for real this time)
closed 2018-08-10 by stale[bot]