iNavFlight/inav #3200: Servo Mixer Doesn't Save in iNav 1.9.3 Configurator

Zangetsu57, 2018-05-12:
Hello,

I used the servo mixer in the iNav 1.9.2 configurator and iNav 1.9.0 firmware.  It worked great.  I just upgraded to the iNav 1.9.3 configurator and 1.9.1 firmware.  The servo mixer no longer saves.  Additionally, when I type a smix in CLI, it doesn't work.  Here's an example command that I'm using in CLI for a 10% pitch, 90% roll mix (I wouldn't use this in practice, it's just an extreme example for testing).  The servos are on a flying wing on servo channels 3 and 4.

smix 0 3 5 10 0
smix 1 4 5 10 0
smix 2 3 4 90 0
smix 3 4 4 -90 0

Thank you for any assistance.

AnandNL, 2018-05-12:
That happens when there already is a mix stored. use reset option first then it will save.

Zangetsu57, 2018-05-12:
I tried the smix reset command, saved, and using the servo mixer in the GUI and it still didn't work for RC Pitch and RC Roll.  When I rolled back to the 1.9.2 configurator and 1.9.0 firmware it works fine.  Maybe I'm still doing something wrong, but did sit there for a couple hours trying different things to get it to work with the latest configurator and firmware.

advancetom, 2018-05-13:
Custom Airplane chosed

advancetom, 2018-05-13:
I tried both custom and normal airplane , but smix in cli dont show values ,same in configurator go to servo tab and normaly you see what mixer is set also nothing.Whats wrong?

DzikuVx, 2018-05-13:
Just tested with Configurator 1.9.3 and FC 1.9.1 and everything is working correctly.
Select mixer airplane custorm, save and reboot, go to Servos tab, configure servos, save. Disconnect, connect, everything is there:

[image]
[image]

DzikuVx, 2018-05-13:
In an example above, smix was configured only using Servo tab. No CLI commands

Zangetsu57, 2018-05-13:
I had my mixer set to flying wing.  Could that have been the issue?

DzikuVx, 2018-05-14:
In 1.9 `smix` can be set only for CUSTOM and CUSTOM_AIRPLANE

closed 2018-05-14 by DzikuVx