iNavFlight/inav #3201: Display problem with OSD

flashez, 2018-05-12:
### Board and Version
It's an Omnibus F3 (with sd), probably pro version because it has barometer.
# version
# INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c1922)

Getting osd problems with some parameters such as:
- voltage (and cell voltage): displaying weird thing
- flymode displaying only: 100FS100

DVR: https://www.youtube.com/watch?v=6tVS7ITA2r4

#### Blackbox logs/Config (.ZIP) where problem occurs
https://pastebin.com/1xHseGdx

If you have any idea of what is occuring ;)

flashez, 2018-05-12:
Just found this similar issue: https://github.com/iNavFlight/inav/issues/3135
Gonna try to update the font

flashez, 2018-05-12:
Ok so basically this is written on the main page: 

> Always load OSD fonts when starting using a new board from the configurator OSD tab to avoid artifacts on the video feed.

rcabuser, 2018-05-12:
Great thx, that really helped me:-) my OSD screen where covered with the letter "V". did the font upload and now it works great, thx again bro:-)