iNavFlight/inav #3212: Led problem using Matek F405 STD

rcabuser, 2018-05-13:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. Matek F405 STD
_Version of INAV used goes here_  1.9.1

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Not getting led strip working correctly, weird and different pattern every time i power up the aircraft. Used a cheap china board earlier (Omnibus) where the led's worked perfectly.

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

closed 2018-05-16 by DzikuVx

rcabuser, 2018-05-16:
Yeah, i'm using the S5 for led.

gstyle97, 2018-05-17:
Me too, on s5 don't work led.

rcabuser, 2018-05-17:
Yeah ok, damn irritating:-(

MATEKSYS, 2018-05-17:
Probably there is sensitivity difference from different 2812LED manufacturer.
Tested it one of our LED strips just now. works fine.
If your LED strip doesn't work. try to solder LED DIN here, or bridge 2 sides of this resistor.

[image]

MATEKSYS, 2018-05-17:
F405-CTR

[image]

rcabuser, 2018-05-18:
Yeah ok, thx for the input. Too bad my board is mounted/buried in my plane now, strange that the led strips i bought worked in my first cheap clone board without a problem.

holoduke, 2019-01-18:
anyone knows this for the matek F405 WING. I have big issues with bad signal while using long wires (1 meter). I don't know where this resistor is on that board.