iNavFlight/inav #3226: Possible (display?) problem with NOTCH filters.

Lord-Quake, 2018-05-16:
According to the diagram raw Gyro goes through LPF and then through Notch 1....
[image]

However this is what I'm seeing (INAV 1.9.1 and INAV Configurator 1.9.3):  
[image]
Both have the same values. Even though Notch 1 is not active it should at least have taken LPF (in this case GYRO LPF cutoff is set to 25Hz) into account or am I missing something?

DzikuVx, 2018-05-16:
This is how it looks in the firmware:

I hope it answers your question

Paweł Spychalski

Od: Leo
Wysłano: środa, 16 maja 2018 21:46
Do: iNavFlight/inav
DW: Subscribed
Temat: [iNavFlight/inav] Possible (display?) problem with NOTCH filters.(#3226)

According to the diagram raw Gyro goes through LPF and then through Notch 1....

However this is what I'm seeing (INAV 1.9.1 and INAV Configurator 1.9.3):

Both have the same values. Even though Notch 1 is not active it should at least have taken LPF (in this case GYRO LPF cutoff is set to 25Hz) into account or am I missing something?

Lord-Quake, 2018-05-17:
Pawel, I'm sadly seeing nothing :-(
Did you insert a picture or a code snippet in your post?

DzikuVx, 2018-05-17:
An image and I see it was sent

Paweł Spychalski

Od: Leo
Wysłano: czwartek, 17 maja 2018 06:44
Do: iNavFlight/inav
DW: Paweł Spychalski; Comment
Temat: Re: [iNavFlight/inav] Possible (display?) problem with NOTCH filters.(#3226)

Pawel, I'm sadly seeing nothing :-(
Did you insert a picture or a code snippet in your post?

Lord-Quake, 2018-05-17:
Hmmmmm....
[image]

DzikuVx, 2018-05-17:
[image: Przechwytywanie2.PNG]

Lord-Quake, 2018-05-17:
Still no go :-(
[image]

Could you send it to my email leo@nutz.de ? Thanks!

Lord-Quake, 2018-05-17:
Got it :-)

[image]

closed 2018-05-18 by DzikuVx