iNavFlight/inav #3231: Omnibus F3/F4  (all versions) servo pin# output for Pitch control 2axis Gimbal/BGC2.2 controller

alanfj, 2018-05-17:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

alanfj, 2018-05-17:
I have a Omnibus F3 v1.1 and a Omnibus F4 v2 and do not know what pin #'s for servo output to use. I have tried motor pins 7&8 on my F3 and motor 5&6 on my F4. Help?

DzikuVx, 2018-05-18:
You have to wait until INAV 2.0 to make that work on those boards

alanfj, 2018-05-18:
Thanks for your quick response & heads up! As usual, "You're the Man!" Thanks again. PS: the neweast version, INAV 1.91, is working very well, too.

alanfj, 2018-05-18:
I'll be patiently waiting. I've got four aerial photography drones w/ these FC's & BGC2.2b camera gimbals waiting for this super upgrade addition.

closed 2018-07-04 by DzikuVx

Kerleston, 2018-07-30:
I am trying Inav 2 RC4 with Omnibus F3 and Motor 5 and 6 is not assigned in resource map.
I could not do the remap.