iNavFlight/inav #3232: FR : Multiple functions on the same LED ?

LemarinelNet, 2018-05-18:
In LibrePilot, LED setup is more simple : 
-Up to 32 led are all set with the same function, and are used to indicate the copter status : 
White blinking, OK (standby)
Green short blinking : GPS Ready
etc....

It would be nice if we could have a similar feature : I have only one LED on my MC, and I really like the GPS status function (red, waiting, green, OK Fix !), but I also like the ARM/Disarm status , etc...

BTW, I'd like to help in translating the wiki in french (and other things ?)... How can I help ?

shellixyz, 2018-05-21:
> BTW, I'd like to help in translating the wiki in french (and other things ?)... How can I help ?

I think a new project needs to be added for a separate wiki in French. Like for the one [in Russian](https://github.com/iNavFlight/ru_wiki/wiki).
The easiest I think would be to clone the wiki repository.

`https://github.com/iNavFlight/inav.wiki.git`

I can help for the translation. I want to translate the configurator GUI too at some point.

LemarinelNet, 2018-05-29:
What about the LED function ?
LibrePilot LED function is really cool ;)

closed 2022-05-07 by DzikuVx