iNavFlight/inav #3242: Matek F722 target does not compile

thegreen4me, 2018-05-19:
christian@Asus:~/inav$ make TARGET=MATEKF722
/home/christian/inav/src/utils/compiler.rb:85:in `run': Compiler error: (RuntimeError)
/usr/bin/arm-none-eabi-g++ -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsingle-precision-constant -Wdouble-promotion -flto -fuse-linker-plugin -Os -I./src/main -I./src/main/target -I./lib/main/STM32F7xx_HAL_Driver/Inc -I./lib/main/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I./lib/main/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I./lib/main/CMSIS/CM7/Include -I./lib/main/CMSIS/CM7/Device/ST/STM32F7xx/Include -I./src/main/vcp_hal -I./lib/main/FatFS -I./lib/main/MAVLink -I./src/main/target/MATEKF722 -ggdb3 -DDEBUG -std=c++11 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -Werror=switch -ffunction-sections -fdata-sections -DSTM32F722xx -DUSE_HAL_DRIVER -D__FPU_PRESENT -DHSE_VALUE=8000000 -DFLASH_SIZE=512 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DMATEKF722 -D__FORKNAME__=inav -D__TARGET__=MATEKF722 -D__REVISION__=de00fc5 -dM -E -o ./obj/stamp ./obj/stamp.cpp
arm-none-eabi-g++: error: unrecognized argument in option '-mcpu=cortex-m7'
arm-none-eabi-g++: note: valid arguments to '-mcpu=' are: arm1020e arm1020t arm1022e arm1026ej-s arm10e arm10tdmi arm1136j-s arm1136jf-s arm1156t2-s arm1156t2f-s arm1176jz-s arm1176jzf-s arm2 arm250 arm3 arm6 arm60 arm600 arm610 arm620 arm7 arm70 arm700 arm700i arm710 arm7100 arm710c arm710t arm720 arm720t arm740t arm7500 arm7500fe arm7d arm7di arm7dm arm7dmi arm7m arm7tdmi arm7tdmi-s arm8 arm810 arm9 arm920 arm920t arm922t arm926ej-s arm940t arm946e-s arm966e-s arm968e-s arm9e arm9tdmi cortex-a15 cortex-a5 cortex-a7 cortex-a8 cortex-a9 cortex-m0 cortex-m0plus cortex-m1 cortex-m3 cortex-m4 cortex-r4 cortex-r4f cortex-r5 cortex-r7 ep9312 fa526 fa606te fa626 fa626te fa726te fmp626 generic-armv7-a iwmmxt iwmmxt2 marvell-pj4 mpcore mpcorenovfp native strongarm strongarm110 strongarm1100 strongarm1110 xscale
arm-none-eabi-g++: error: unrecognized argument in option '-mfpu=fpv5-sp-d16'
arm-none-eabi-g++: note: valid arguments to '-mfpu=' are: crypto-neon-fp-armv8 fp-armv8 fpv4-sp-d16 neon neon-fp-armv8 neon-fp16 neon-vfpv4 vfp vfp3 vfpv3 vfpv3-d16 vfpv3-d16-fp16 vfpv3-fp16 vfpv3xd vfpv3xd-fp16 vfpv4 vfpv4-d16
	from ./src/utils/build_stamp.rb:66:in `hash_config'
	from ./src/utils/build_stamp.rb:44:in `stamp'
	from ./src/utils/build_stamp.rb:87:in `<main>'
Makefile:983: recipe for target 'obj/build.stamp' failed
make: *** [obj/build.stamp] Error 1

shellixyz, 2018-05-19:
Your compiler is probably too old. What version ?

thegreen4me, 2018-05-20:
Yea works with an updated compiler. Turns out the one in the debian 8 repo is almost a decade old!! what a joke.

closed 2018-05-20 by thegreen4me