iNavFlight/inav #3248: Disarmed /Armed in flight

wiesiekkr, 2018-05-20:
I welcome colleagues,
today during the flight the jumped out inscription ARMED and the position Home was reset :(
Good that I knew the area otherwise I would not get it home. Armed/Disarmed is by Switch
https://www.youtube.com/watch?v=4G8lT5wSW9Q

# 
[dff 1.9.1 po autotrym.txt](https://github.com/iNavFlight/inav/files/2020532/dff.1.9.1.po.autotrym.txt)
`

fiam, 2018-05-20:
@wiesiekkr It's a known bug in going back to at least 1.7.2, but it only started happening lately. We're not 100% sure about what caused the sudden increase in reports, but the good news is that it's fixed for 2.0. See #2981, where it was originally reported. I'm closing this one as a duplicate.

fiam, 2018-05-20:
Ouch, spoke too fast. I've just seen you're using ELERES over SPI, so #2981 doesn't apply.

DzikuVx, 2018-05-21:
I think it #3118 applies over here

wiesiekkr, 2018-05-21:
Thx Pawel, 
The second thing is jumping GPS in 1:01min on video. I take the height measurement from the BARO and this situation happens often :(

digitalentity, 2018-06-30:
@DzikuVx we might have introduced a possible issue when we allowed `rcData` to be updated while receiver is in failsafe state a few releases ago. I'm working for a fix to have a) possibility to display rc channels in Configurator regardless or failsafe status and b) don't process `rcData` when receiver is reporting signal loss.

flounderscore, 2018-07-08:
I had two cases of mid-air disarm ("Disarmed by switch") flying my mini drak yesterday. 

First, I had a successful flight without using the launch mode (BB_SUCCESS.zip) and activated launch mode for the next flight. 

If I remember correctly, very soon after I took manual control in the second flight the FC disarmed mid-air. I could get control back by toggling the arming switch (although I still had the summary screen on the osd) and landed dead-stick. I couldn't spin up the prop again, because I didn't remember to how to do it with launch-mode activated. (BB_DISARM.zip)

At that point I was convinced I must have accidentally bumped the arm switch and gave it another shot. Launch worked fine, but, again, after a short time (felt a little longer this time) the FC disarmed again mid air and this time I could not regain control. Luckily, the plane was recovered with minor damage from a swampy area near the field.

Previous to the three flights I was running iNav 1.7 and had just upgraded to the latest git-version (see DIFF.txt, did full chip erase. I also did not copy the old CLI config, but reconfigured by hand). The XSR RX might have an issue, because the RSSI seems to be lower than a R-XSR on another aircraft, but as you can see in the logs, never really drops too low. This other aircraft is flying without issues with a slightly earlier GitHub version (main difference: no GPS).

There is certainly a chance I forgot to change some settings. Maybe you guys can see something in the attached logs. Also, I feel like the GPS track in the log ends too soon and I had control for another second or so - just a feeling, unfortunately, I did not record a video

[BB_DISARM.zip](https://github.com/iNavFlight/inav/files/2174188/BB_DISARM.zip)
[BB_SUCCESS.zip](https://github.com/iNavFlight/inav/files/2174189/BB_SUCCESS.zip)
[DIFF.TXT](https://github.com/iNavFlight/inav/files/2174190/DIFF.TXT)

digitalentity, 2018-07-16:
Since https://github.com/iNavFlight/inav/pull/3482 is merged, RC receiver failsafe settings won't mess up processed channel values (`rcData`). I think this issue should now be fixed.

digitalentity, 2018-08-05:
Closing. Issue with spontaneous disarm seems to be resolved

closed 2018-08-05 by digitalentity