iNavFlight/inav #3256: Adjust Pitch/roll from CMS

povlhp, 2018-05-22:
Please as an option to adjust pitch/roll from the CMS, so we can do it in the field.

For horizon modes, finding the optimal pitch for long glide is a process, and not something you can adjust perfectly home at the desk, so we need a way to adjust it thru CMS.

I would consider this a next step from #3169. https://github.com/iNavFlight/inav/pull/3169

shellixyz, 2018-05-22:
Not saying we can't add pitch/roll alignment to the CMS but you can already adjust them in flight using the RC adjustments since 1.9 (#2768)

shellixyz, 2018-05-22:
Yes you can use the stick commands on the ground to adjust the pitch/roll alignment too.

povlhp, 2018-05-22:
Then just close this request. 
And document it in wiki. It is an important one to be able to do in the field to get things fully trimmed.

shellixyz, 2018-05-22:
Indeed it is important. It is documented here: https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

povlhp, 2018-05-22:
Idea for text:
https://github.com/iNavFlight/inav/wiki/Fixed-wing-guide

If the plane changes pitch, often more up, when going from manual mode to horizon mode, then the FC pitch is off, typically because wings and where you mount the FC are not oriented perfectly. To fix this, you can land the plane, disarm, and then use stick commands listed here: https://github.com/iNavFlight/inav/blob/master/docs/Controls.md - The Trim Acc commands are what you are looking for. Every time you press the right stick up, the pitch will decrease by 0.1 degrees, allowing you very fine grained control. It is not unusual to have a pitch 4-6 degrees off. That is 40-60 "clicks" with the stick. When done, use the stick command to save settings. To go back, power off FC before saving. Use same method to fix roll.

The perfect pitch is where you fly manual and level at your normal speed (often 40-60% throttle), and don't see any pitch change when switching to horizon mode. The pitch should be adjusted such that an unpowered plane will descend and keep an airspeed that is sufficient above stall speed to allow safe maneuvering, and allow unpowered landing.

closed 2018-08-04 by stale[bot]