iNavFlight/inav #3263: Custom Servo Mix, no Servo movement at all.

janmaier27, 2018-05-22:
Hey,
i hate the standard servo mix for fixed wing since the maximum roll is low it also wont give full servo throw even on passthrough and maximum stick input. Tried to create a custom mix and in the motor tab it acts correctly, but as soon as i power up my wing servos wont move, even the usual initial centering wont take place.
Its probably my own dumbness but i cant figure out whats wrong. As soon as i go back to standard fixed wing setup it works again.
Im using an Omnibus F4 Pro.
CLI:
# version
# INAV/OMNIBUSF4PRO 1.9.0 Mar  5 2018 / 00:15:43 (dc4ef594f)

# resources

# mixer
mixer CUSTOM

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 1 100 0
smix 1 3 0 100 0
smix 2 4 1 100 0
smix 3 4 0 -100 0

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1230 1770 1500 -100 -1 
servo 4 1230 1770 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature MOTOR_STOP
feature GPS
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
[... 506 lines cut ...]
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 0
set thr_expo = 35
set tpa_rate = 0
set tpa_breakpoint = 1350
set rc_expo = 35
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 35
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

shellixyz, 2018-05-22:
Use `mixer CUSTOMAIRPLANE` it should work but the default flying wing mixer is good. You will get maximum deflection by using pitch+roll.

DzikuVx, 2018-05-23:
Just like @shellixyz wrote, default flying wing mixer is good: https://www.youtube.com/watch?v=o2RwTeBbvos

in 1.9 `CUSTOMAIRPLANE` is what you are looking for if you really need, but I advise against for the health of the servos

closed 2018-05-23 by DzikuVx

janmaier27, 2018-05-23:
Actually the Standard Mixer will never output more than 1850 or less than 1150 even on full roll+pitch. Im used to the standard Taranis fixed wing mix, and compared to that iNavs Max roll is super low and max servo throw is reduced. Is it supposed to be that way? Thanks for the tip, will try customairplane!

DzikuVx, 2018-05-23:
@janmaier27 1850-1150 is true for motor, not servo and only on default values. Min/Max servo output can be adjusted in Configurator Servo Tab for every servo independently

janmaier27, 2018-05-23:
@DzikuVx Even if i set the servos to full range it wont give me full range unless i use my custom Mix.