iNavFlight/inav #3267: Compass abruptly starts producing erroneous but realistic looking data when in flight

rgregs, 2018-05-23:
iNav Version: 1.9.0

Boards: Omnibus F4 V3 pro

Problem: Compass abruptly starts producing erroneous but realistic looking data a few mins into flight.

450 frame quad, compass (with integrated M8N GPS on raised platform, https://hobbyking.com/en_us/ublox-micro-m8n-gps-compass-module-1pc.html) is calibrated and consistent during flight. Home arrow is correct regardless of reasonable changes in attitude and RTH works. And then, at some random time during flight, the compass looses correct heading, the home arrow is completly wrong/stuck. 

[LOG00011.TXT.ZIP](https://github.com/iNavFlight/inav/files/2030545/LOG00011.TXT.ZIP)

Blackbox logs show mag_Z instantly changing in flight at 4:58 from about -550 to about +25, with X and Y also instantly changing. After the abrupt change, mag_XYZ continue to produce gently varying numbers that look real but are completely different values to before and consequently produce a wrong heading.

Power cycle the quad and the mag values are correct again. Power cycling the mag+gps, the reported values look approximately correct but no longer update.

Obviously this problem breaks RTH and failsafe-RTH and would have lead to a fly-away if I'd not had RX signal at the time. The quad was flying in pos hold, fortunately stick input is more powerful than pos hold.

This problem can also be seen on the test bench. F4 powered by USB to laptop running Configurator, F4 powering RX and GPS/Mag, TX off, VTX and ESCs off. Leaving the quad leaning on a ~+20 degree roll, unarmed, facing east, the Configurator shows ~90 degree heading and changes a small amount over time, as expected. After (in one case)25 mins, I find the quad in the Configurator is facing north, heading 4 dgrees. By this point there were 9 i2c errors, they had build up steadily over the 25 mins. The same problem was seen with the quad in that position over night, in this case there were 59 i2c errors by the morning and it took at least 2 hours (the last time I looked before going to bed) for the compass problem to appear.

Have tried reducing i2c speed to 100KHz, problem still appears. The bench tests above were at 100KHz, the blackbox log during flight was at 400KHz. 800KHz basically doesn't work, Configurator shows MAG coming and going with every update. The mag needed pull-up resistors to get it working at all.

Update: uCenter has been used to reset M8N output messages to stop GPS sat count blinking to 0. Also recalibrated mag, producing similar but different magZero numbers, as expected. The compass output remained correct for a few hours and again overnight. However, by the afternoon of the next day, the compass heading was wrong. Power cycling the F4 by removing USB, the compass was producing a bad heading within an hour.

Config dump:
[inav.dump.zip](https://github.com/iNavFlight/inav/files/2030569/inav.dump.zip)
(mostly defaults, have reduced PIDs and some LPF values to suit the larger frame, lowering Angle LPF was necessary to avoid Angle mode wobble. These PIDs are functional but soft and not optimal).

zipray, 2018-05-23:
Be aware of the time of geomagnetic storm released by the observatory.

teckel12, 2018-05-24:
@zipray You lose all credibility when you start blaming "cosmic rays" 😉