iNavFlight/inav #3270: INAV 1.8 no gps arming issue

JDFPV, 2018-05-23:
I have an omnibus f4 pro board, with no gps. I have no gps selected under the configuration tab, nor any nav flight modes. I can not get this board to arm using 1.8. I do have a arm switch. Please help

shellixyz, 2018-05-23:
Please post config dump and output of the `status` command

shellixyz, 2018-05-23:
`nav_extra_arming_safety = OFF` is not needed if none of the navigation modes are used.

JDFPV, 2018-05-23:
Thats it. Thanks

closed 2018-05-23 by JDFPV

digitalentity, 2018-05-24:
@shellixyz problem might be that firmware doesn't report and Configurator doesn't display modes that are lacking sensors. If you disable GPS in Config you won't see RTH mode, but it still will be configured (this is something we could improve).

digitalentity, 2018-05-24:
@JDFPV great that you fixed that.

shellixyz, 2018-05-24:
Indeed might be the reason that's why I wanted to see a dump. How could we improve that ? Show the modes when they are configured even if they should not be shown because of lacking hardware ?

digitalentity, 2018-05-24:
Good idea. We can also light up those modes in red color in Configurator to grab user attention.