iNavFlight/inav #3271: INAV 1.8 Tricopter / no servo movement

JDFPV, 2018-05-24:
Im not getting any servo movement from the omnibus f4 pro board. Using the default preset for the 280 tricopter. Ive tried everything but i can not get it to work. Pleaae help

shellixyz, 2018-05-24:
Please post a config dump

JDFPV, 2018-05-25:
# version
# INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

# resources

# mixer
mixer TRI

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature CURRENT_METER
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 476 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 40
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1650

#

shellixyz, 2018-05-25:
Where did you connect the servo ?

JDFPV, 2018-05-25:
I tried connecting the servo in all the available slots. Minus the motor slots. All 3 motors work fine with the board. Just no servo

shellixyz, 2018-05-25:
If you didn't enable the camera gimbal feature (SERVO_TILT) it should be connected to output 5. Don't forget that the 5V rail has no power by default you need to supply it with a BEC.

JDFPV, 2018-05-26:
I did not enable camera gimbal. But i am using a 5v bec to power the servo. Let me test and see if that will work.

JDFPV, 2018-05-26:
I enabled sevro gimbal. And it still does not work. What else can i try ?

shellixyz, 2018-05-26:
Enabling servo gimbal won't make it work. I mentioned it because when you enable it the first 2 servo outputs are dedicated to the gimbal servos so it offsets the "normal" servo outputs by 2 positions. You can't use the servo gimbal feature with this board for a tri since it only has 2 servo outputs. If you enable it there is no space left for the yaw servo.

> What else can i try ?

I don't know. The output 5 should work I'm using it in a plane.

DzikuVx, 2018-06-13:
INAV 2.0 brings visual information which output is assigned to which servo as well and deprecates SERVO_GIMBAL and C HANNEL_FORWARDING. We can close this one I suppose

closed 2018-06-13 by DzikuVx