iNavFlight/inav #3273: OSD CLI != GUI

Artyum, 2018-05-25:
Hi, In Configurator 1.9.3 and omnibus F4 v3 I try to setup OSD by using CLI. I noticed that after I run some commands eg:
set osd_home_dir_pos = 720
set osd_home_dist_pos = 3021
I can't see the corresponding attributes in OSD Preview window  anymore.  I've disabled everything else beside above two and they are not there.  I can click on and off switch for Direction and Distance to Home but still no icons in the preview.
Best Regards

shellixyz, 2018-05-25:
That's because you put them outside of the display area. Do this and you'll see them again (second column, second row):

```
set osd_home_dir_pos = 66
set osd_home_dist_pos = 66
```

Artyum, 2018-05-26:
That was it, thank you.

closed 2018-05-26 by Artyum