iNavFlight/inav #3279: Matek 405 CTR upside down installation

DimychK, 2018-05-26:
###  

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Matek-405CTR
_1.9.1_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* Due to frame compatibility I installed board upside down, in settings I set roll 180. On a picture everything looks good, but the motor sequence remains inverted and in a few seconds after arming quad made a flip itself. What’s wrong? Thank You
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

shellixyz, 2018-05-26:
How did you wire the motors ? Which motor on which output ?

DimychK, 2018-05-27:
Motors are wired exactly as stated on the board, 1 to 1, etc. and it looks normal, when board is positioned in normal mode. When the board is positioned upside down it looks illogical to wire the motors across, so I expected, when roll of 180 degrees is applied, the motor numbers would be repositioned accordingly, 1 and 3 would be swapped, and 2 and 4 as well. If I wrong, is there any way to change the motors numbers without rewiring?

digitalentity, 2018-05-27:
Use Configurator motor tab to verify that motors are wired correctly: 1 =
rear right, 2 = forward right etc. Regardless of how you rotate the board
motor 1 (rear right) should be wired to M1 output etc.

shellixyz, 2018-05-27:
@DimychK You can wire the motors as you like. You just need to update the motor mixing accordingly. Outputs aren't swapped automatically when you change board orientation.

DimychK, 2018-05-27:
Thank you!Could you tell me the way how to do that? Appreciate it.

Best regards
Dmytro

shellixyz, 2018-05-27:
The default mixer for a quad is:
```
mmix 0 1 -1 1 -1
mmix 1 1 -1 -1 1
mmix 2 1 1 1 1
mmix 3 1 1 -1 -1
```

You need to select custom mixer in the configuration tab then enter your modified motor mixer in the CLI. Default motor order is rear-right, front-right, front-left, rear-left. So if you want change the outputs for 180 roll you need to swap left and right motors, this should work:
```
mmix 0 1 1 -1 -1
mmix 1 1 1 1 1
mmix 2 1 -1 -1 1
mmix 3 1 -1 1 -1
```

You can find more information on motor mixing here: https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md

DimychK, 2018-05-28:
Thanks. I’ll try today.

DimychK, 2018-05-28:
Well, something is still wrong. Immediate flip after arming. I’ve  checked parameters twice.
I couldn’t understand what does it mean in docs positive (negative) roll.? Anyway it is a way to solve the problem I have, I’ll play with it

shellixyz, 2018-05-28:
The mmix parameters describe how the FC should raise/lower the corresponding motor RPM depending on the requested roll/pitch/yaw angular velocity. If your copter flips it means it does the inverse of what it should do. For example if it does a roll flip it means the roll parameter for the left and right motors are inverted compared to what they should be. Rolling right means the left motors should spool up and the right motors should spool down so the roll mmix parameter for the left motors should be -1 and for the right motors should be 1. Same logic for pitch. For CW yaw the CW motors should spool down and the CCW motors should spool up so the yaw mmix parameter for rear-right and front-left motors should be -1 and for and rear-left and front-right should be 1.

I mixed up the default motor order. It is rear-right, front-right, rear-left, front-left so the right mmix corrected for 180 roll is:
```
mmix 0 1 1 1 1
mmix 1 1 1 -1 -1
mmix 2 1 -1 1 -1
mmix 3 1 -1 -1 1
```

DimychK, 2018-05-28:
Thank you!
I’ll try again tomorrow

DimychK, 2018-05-28:
A special thank for the explanation. It quite difficult to find out detailed explanation how does it work. I’m good with physics and aerodynamic (professionally), but board programming - it is something different. It is good sometimes to understand exactly what am I doing. Thank you.

DimychK, 2018-05-30:
Yeah! It flies! Thank for help!

closed 2018-05-30 by DimychK