iNavFlight/inav #3280: Omnibus F3 v1.1 Hexacopter Motor 5&6 pin output problem

alanfj, 2018-05-26:
I am building a F550 Hexacopter with an Omnibus F3 v1.1 FC board. I flashed it with the latest version of INAV 1.91. My problem is, I can not get motors 5 & 6 to work. I tried ESC pins 5 & 6, and also 7 & 8. Is there something I can do to fix this? Help!

shellixyz, 2018-05-26:
Please post a config dump

alanfj, 2018-05-26:
# dump

# version
# INAV/OMNIBUS 1.9.1 Apr 21 2018 / 12:53:03 (03a5c1922)

# resources

# mixer
mixer HEX6X

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 503 lines cut ...]
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 50
set thr_expo = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

#

alanfj, 2018-05-26:
Hope the above helps.

shellixyz, 2018-05-26:
Outputs 5 and 6 on this board are shared with UART3. You can't use both at the same time. If you want to use outputs 5 and 6 you need to disable everything on UART3 in the ports tab.

alanfj, 2018-05-26:
So, if I have a PPM/IBUS or a Spektrum receiver which is sharing  UART 3, what receiver should I use instead for the hexa-copter? I have a multi-protoc;al transmitter (Walkera 12).  Any suggestions? What other FC would fulfil my needs? I would also like a FC that would work with a BDC 2.2 camera gimbal(pivot& yaw).

shellixyz, 2018-05-27:
> So, if I have a PPM/IBUS or a Spektrum receiver which is sharing UART 3, what receiver should I use instead for the hexa-copter?

You can use a PPM receiver. It would free up an UART. So you wouldn't need to use UART3 I guess.

> What other FC would fulfil my needs? I would also like a FC that would work with a BDC 2.2 camera gimbal(pivot& yaw).

You can use any FC with enough broken out UARTs for your needs. The MATEKF405 for example has 7 outputs and 7 UARTs none shared with the outputs. Any FC would work if your gimbal takes PWM as input as soon as it has enough servo outputs.

alanfj, 2018-05-27:
OK, I removed my Spectrum serial receiver, releasing UART3, then removed SBUS jumper on OMNIBUS F3 v1.1, and placed a PPM receiver on it and it all works now! :)  Thanks for your help!

closed 2018-05-27 by alanfj