iNavFlight/inav #3284: Multiple blackbox logs.

NeilGaede, 2018-05-27:
It would be nice to have a black box log for each arming event. Right now I am only getting one log per power up. If I fly around a bit and then land to make an adjustment, anything after the first disarm is ignored by black box. 

Thanks.

shellixyz, 2018-05-27:
Please post a config dump.

digitalentity, 2018-05-27:
Please upload a blackbox log file when you did multiple arming events and a config dump.

NeilGaede, 2018-05-28:
Entering CLI Mode, type 'exit' to return, or 'help' # dump

# version
# INAV/MATEKF405 1.9.1 Apr 21 2018 / 13:25:29 (03a5c1922)

# resources

# mixer
mixer CUSTOM

mmix reset

mmix 0 1.000 -0.879 1.000 1.000
mmix 1 1.000 -0.879 -1.000 -1.000
mmix 2 1.000 0.879 1.000 -1.000
mmix 3 1.000 0.879 -1.000 1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1

# feature
feature -RX_PPM
feature -VBAT
feature -TX_PROF_SEL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
[... 504 lines cut ...]
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_mc_heading_p = 60
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set nav_fw_heading_p = 60
set thr_mid = 0
set thr_expo = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
set fw_tpa_time_constant = 0
set rc_expo = 0
set rc_yaw_expo = 0
set roll_rate = 30
set pitch_rate = 30
set yaw_rate = 30
set manual_rc_expo = 0
set manual_rc_yaw_expo = 0
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

stronnag, 2018-05-28:
loop = 125 is probably the reason. Does logging work at a more conventional iNav loop (1000, 500) ?

NeilGaede, 2018-05-28:
If I set logging at 1/8 the time is longer but the file size remains the same 4mb max.

NeilGaede, 2018-05-28:
I was able to get larger logs by turning down PID loop to 1k. It seems to work up to 4k, with logging at 100%. Not sure why it won't handle 8k, but I guess this is ok for now. It would be really nice to be able to upload logs using INAV configurator, though, without having to pull the SD card out every time. :) I've been able to stop and start logs using a switch, but if I just leave it at arm/disarm it only takes the first pair of events.

closed 2018-08-10 by stale[bot]