iNavFlight/inav #3286: INav 1.9.1 not arming in POSHOLD mode

Seeelefant, 2018-05-27:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
MATEK F722
inav 1.9.1

_Use CLI command `version` and put its output here_
# version
# INAV/MATEKF722 1.9.1 Apr 21 2018 / 13:43:00 (03a5c1922)
_If it's a custom-compiled firmware please mention this_

### Behavior
Switching to mode "POSHOLD" leads to "Navigation is unsafe"
POSHOLD
 
Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 216 seconds
Current Time: 2018-05-27T14:49:14.041+00:00
Voltage: 16.43V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x20010000, Heap available: 1024
I2C Errors: 0, config size: 3500, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 2004, PID rate: 499, RX rate: 110, System rate: 9
Arming disabled flags: NAV CLI

#
# bootlog
Time Evt            Description  Parameters
   1:  0          CONFIG_LOADED 
   1:  1       SYSTEM_INIT_DONE 
 257: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 257: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 257: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 257: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 257: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 257: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 257:  2          PWM_INIT_DONE 
 907:  9         GYRO_DETECTION  (7, 0, 0, 0)
1398: 10          ACC_DETECTION  (8, 0, 0, 0)
1499: 11         BARO_DETECTION  (4, 0, 0, 0)
1499: 20        PITOT_DETECTION  (0, 0, 0, 0)
1509: 12          MAG_DETECTION  (2, 0, 0, 0)
1620:  4       SENSOR_INIT_DONE 
2125:  5          GPS_INIT_DONE 
2125:  7    TELEMETRY_INIT_DONE 
3126:  8           SYSTEM_READY
# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       5       1    0.5%    0.5%         3
 1 -          PID     499     253     125   13.1%    6.7%     19498
 5 -           RX     111      58      39    1.1%    0.9%      1386
 6 -       SERIAL      99     483       3    5.2%    0.5%       374
 7 -      BATTERY      49   40050       9  196.7%    0.5%       174
 8 -  TEMPERATURE       1       2       1    0.5%    0.5%         0
 9 -       BEEPER      99      14       3    0.6%    0.5%       110
10 -          GPS      24     169      42    0.9%    0.6%       316
11 -      COMPASS       9     231     220    0.7%    0.6%       678
12 -         BARO      43     250     165    1.5%    1.2%      2197
15 -    TELEMETRY     500     138       4    7.4%    0.7%      1200
18 -          OSD     249     146      16    4.1%    0.8%      1416
19 -          CMS      49      12       1    0.5%    0.5%        25
22 -      VTXCTRL       4       6       2    0.5%    0.5%         1
Task check function            19       4                       147
Total (excluding SERIAL)                   228.1%   14.5%

_Is this a new feature request?_
No
#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

Seeelefant, 2018-05-27:
Well, not that easy, even without CLI it is not arming.

shellixyz, 2018-05-27:
You can't takeoff in POSHOLD mode it is unsafe that's why you can't arm when it is enabled.

DzikuVx, 2018-05-27:
This is by design. We had this discussion multiple times. You can not arm if any NAV mode (poshold, rth, etc) is enabled

closed 2018-05-27 by DzikuVx