iNavFlight/inav #3288: Compas AK8975 stop working AFTER upgrade from iNav 1.8

robertha64, 2018-05-28:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
Omnibus F3 clone and GPS-compass AK8975 combo both from Banggood
iNav version 1.8 compass IS working, iNav version 1.91 compass NOT working
### Behavior
GPS puck with AK8975 compass module  - compass is working and it is recognised by iNav 1.8 without issue
After flashing any newer version (tried all stable above 1.8) the compass is no longer recognised.

When reflased 1.8 iNav, the compass start working again

Looks like support for AK8975 is droped in new version - at least for Omnibus F3 board.

#### Blackbox logs/Config (.ZIP) where problem occurs

gbine, 2018-05-30:
> Yes, looks like we accidentally removed support for AK8975 from most targets. @gbine<https://github.com/gbine> you test on Sparky or Sparky2 board?

Sparky

gbine, 2018-05-30:
Thanks @digitalentity 
I added (well copied the AK8963 entries and modified) the AK8975 back into common_hardware.c
The developement branch seems to be working now.  
I didn't think to look at common_hardware.c as the correct files seemed to get compiled. Instead I spent a few hours trying to figure out if the compass_ak8975.c code was bad.

One other note...  src/main/target/SPARKY/target.h   the mag alignment is incorrect for my SPARKY. (it is a SPARKY rev2 board)
I needs to be #define MAG_AK8975_ALIGN        CW0_DEG to work correctly by default.

Thanks for your assistance (you pointed me in the correct direction).

digitalentity, 2018-05-30:
@gbine cool, so you fixed it for Sparky! Can you submit a pull-request so we can merge the fix into development?

gbine, 2018-05-30:
Ignore my comment about the alignment.  I will work through it and report back.

gbine, 2018-05-30:
@digitalentity , regarding axis' (X & Y) which axis should point in the forward direction?  and to the right? 
I thought Y+ should be forward, and X+ to the right, with Z+ going skyward.  The info I have for the 9150 (see picture) show that the orientation on the SPARKY is rotated 180 (see picture)

With that information, I expected the ACC and GYRO to be CW180 and the MAG to be CW270FLIP... but that differs from the source tree which has the ACC & GYRO at CW270 and the MAG CW270FLIP

Can you help me figure out where my logic is flawed? 

Thanks

[image]

[image]

digitalentity, 2018-05-31:
X - Forward
Y - Right
Z - Down

gbine, 2018-05-31:
Ahhh. That explains things
Thanks!

gbine, 2018-05-31:
Well, I have tried all combinations of MAG alignment, and nothing works correctly on my SPARKY. I get close but it is either 180 out or rotates when tilted. 
1.8, which does have the AK8975 compiled in, does the same thing for me, so maybe it's an issue with the MPU-9150, but I did try betaflight, and it works correctly with defaults.   We have the same mag_alignments  in src/main/target/SPARKY/target.h as they do.

I'm not sure if it is worth chasing this down.  I use two SPARYs but both in fixed wing so I don't really care, however @robertha64  may; IF it affects his standalone unit.

I will submit a pull request to get the AK8975 code compiled in, but I have never done that before.
@digitalentity can you give me a short tutorial?

shellixyz, 2018-06-01:
@gbine To submit a PR you need to push the changes to a branch in a fork of the iNav repository:

1. Fork iNav on Github
2. Clone your fork: `git clone `https://github.com/gbine/inav`
3. Make a new branch with the development branch as origin: `git checkout -b ak8975_fix development`
4. Edit the files
5. Commit your changes: `git commit -a` and enter a commit message
6. Push your new branch to Github: `git push --set-upstream origin ak8975_fix`
7. Go to your repository on Github, switch to your new branch and click "Pull request" on the right bellow the clone button
8. Switch the base branch to the development branch and click on "Create pull request"

robertha64, 2018-06-12:
To extend my information - to make my standalone AK8975 somehow working I did select CW90 and NO FLIP - that is only selection from 8 variants where at least crucial directions show correctly (after quite time of setling which may indicate some problem). Copter changes heading when tilted (20 deg+) whech probably means the tilt corrections doesnt work. I did repeat mag calibration outside and check the bias and gain changed from default, but results still same.

digitalentity, 2018-06-12:
Most likely in the driver axis directions are mixed up. Sometimes sensor
chip has odd axis orientation. I'll try looking at the code vs datasheet -
maybe I will find some discrepancies.

gbine, 2018-06-12:
I tried to generate a pull request but messed it up, and have been on the road for past 10 days.  I hope to be home by weekend and get the pull request done properly.

robertha64, 2018-06-13:
Attached pictures indicate orientation AKM8975 according datasheet and how chip is installed on the GPS board (on the botom of GPS board).
[image]
[image]

rb1205, 2018-06-22:
I've been having issues on my SP3 evo board mag since 1.9 as well. Do this applies to the MPU9250 chip in any way?

robertha64, 2018-06-22:
MPU9250 has the simmilar magnetometr inside - the AK8963, so it looks is afected as well.

fiam, 2018-06-24:
MPU9250 is not affected by this. There's a specific driver for the compass inside the MPU9250 in compass_mpu9250.c. I'm not sure exactly how this differs from the AK8963 driver, but it seems there are some differences.

fiam, 2018-06-24:
Fixed in #3413

PR #3413 (2018-06-24): Fix support for AK8975 MAG
Supersedes #3324 
Fixes #3288
closed 2018-06-24 by fiam