iNavFlight/inav #3293: Іnteger overflow in the ADC current sensor code

vanessaorgua, 2018-05-29:
In the src/main/sensors/battery.c following code is available

int32_t currentSensorToCentiamps(uint16_t src)
{
    int32_t microvolts = ((uint32_t)src * ADCVREF * 1000) / 0xFFF - (int32_t)batteryConfig()->current.offset * 1000;
    return microvolts / batteryConfig()->current.scale; // current in 0.01A steps
}

range of src variable values - 0x0 to 0xFFF
ADCVREF  eq 3300
let us assume src = 4095
calculation
0x00000FFF * 3300*1000 = 0x3 2577 A560 <- Іnteger overflow here

On my OMNIBUS F4 PRO maximum displayed current is 9,63 Amp ,  if more  -1

maybe this is the best option
int32_t microvolts = ((uint32_t)src * 806UL)  - (int32_t)batteryConfig()->current.offset * 1000;
then there will be no overflow, but there will be less precision

shellixyz, 2018-05-29:
Duplicate of #3124, #3144, #3180 and #3181. Indeed you guessed correctly it is caused by an overflow in this function. It has already fixed in the development branch and a fixed build of 1.9.1 is available:  [inav_1.9.1_current_fix.zip](https://github.com/iNavFlight/inav/files/1955341/inav_1.9.1_current_fix.zip)

Dinirofpv, 2018-05-29:
shelllixyd you can re-compile this OMNIBUS F4 PRO 1.9.1 fix firmware  to correct uarts and inverters to work on Pro Corner board?

vanessaorgua, 2018-05-30:
shellixyz, please show the code for this fix. Thank you.

shellixyz, 2018-05-30:
@vanessaorgua The fix is this: #3131 

@Dinirofpv What do you mean by "fix firmware to correct uarts and inverters to work on Pro Corner board wirh the sbus?" ?

Dinirofpv, 2018-05-30:
@shellixyz i have omnibus f4 pro corner and test 1.9.1 and earlier firmwares and no one off them works the sbus. Test omnibusf4pro and omnibusf4v3 without success. This board have inverter on UART 3 (this port is sheared with ic2) and UART 6 and can be controled by firmware! Once you did a custom firmware for that board firmware 1.8.0 for a user, i found that hex and test and sbus start working in UART 3 but the rgb leds don't work a islittle impossible to make the accelerometer calibration process... you think you can help? I pretend use UART 1 for GPS and softserial to smartaudio (TX M6 pin on the board)...
Sorry for make tou spend time on this
Thanks you

closed 2018-08-12 by stale[bot]