iNavFlight/inav #3295: SPracingF3 Sport telemetry problem INAV above 1.8.0

zmr250cd, 2018-05-30:
Board and Version
SpracingF3, Frsky XSR, Beitian BN880 GPS
Uart 2 : GPS
Uart 3 : SBus
SoftSerial 2 : SPort

With INAV 1.8.0 : All OK, Taranis telemetry screen show "*" for all data (ie : RSSI, BtRx, AccX, GPS, ....)
For INAV 1.9.0 and 1.9.1 Taranis keep shouting "Probe telemetry lost". Only data for RSSI, BtRx and A2 show "*"

teckel12, 2018-06-01:
@zmr250cd If you delete the sensors and then rediscover, does it add the sensors back?  I know it works well on other flight controllers, and I'm almost 100% sure someone has run the SPracingF3 since v1.8.

zmr250cd, 2018-06-01:
I have try this several times : some sensors appears, never the same and a few seconds after, the value of the sensor is between [ ] and taranis says "Probe telemetry lost".....

zmr250cd, 2018-06-01:
Sorry close by mistake !!!!

teckel12, 2018-06-02:
@zmr250cd SPracingF3 clone?  I tried a couple clones and had nothing but trouble.  But had zero issues with actual SPracingF3 FC.  If it's a clone, I'd seriously consider that's the issue.  I no longer buy clone FC, except for mini indoor quads.

zmr250cd, 2018-06-03:
Thanks for your analysis teckel12. I think you are right, my SPracingF3 is a chinese clone... and it's the problem...

closed 2018-06-03 by zmr250cd