iNavFlight/inav #3302: MSP_RTC returning

ShikOfTheRa, 2018-05-31:
Running current 1.9.1 release

MSP_RTC seems to return a unix time which is one day out. 

Have just added support into MWOSD and I get it 1 day out.

Tuesday 29/05/2018 @ 20:10 I get a unix time of 1527711053 from iNav
If I enter that into unixtimestamp.com it also shows it one day later.

It could be GPS error or my implementation - but if I put in simulated data it displays correctly :)

Could someone verify?

stronnag, 2018-05-31:
You appear to be correct, from my ground station log yesterday:
```
2018-05-30T14:07:36+0100  RTC local 2018-05-30T14:07:36.186, fc 2018-05-31T14:07:36.598
```
The ground station (mwp) is on a unix box, so it should handle `time_t` correctly.

However, timestamp is correct in BB logs:

```
mwp log:
2018-05-28T13:35:46+0100 RTC local 2018-05-28T13:35:46.423, fc 2018-05-29T13:35:48.022

BBOX log:
H Log start datetime:2018-05-28T12:54:51.809+00:00
```

stronnag, 2018-05-31:
Not sure how I missed that, obviously just look at the time ...

stronnag, 2018-05-31:
The problem is the way that ` dateTimeToRtcTime()`  tries to account for a leap year; as the base (1970) was in the middle of a a 'leap cycle', it erroneously thinks 2018 (48 years on) is a leap year. So 2018-01-01 and 2018-02-01 give the correct time_t, while 2018-03-01 gives a time_t for  2018-03-02. 

At least that explains why it wasn't spotted when it was introduced last year.

stronnag, 2018-06-01:
Fixed in #3307 
@ShikOfTheRa can you please verify and close

ShikOfTheRa, 2018-06-01:
Thanks stronnag. 
I must admit I thought it was my error for ages went over the code so many times. I do now have  amuch higher appreciation for date calcs!

Have a look at MSP_RTC for betaflight...
https://github.com/betaflight/betaflight/blob/42b2e4077f3b5ad74363557feff632c5a8c2305a/src/main/interface/msp.c#L1314

Doesn't help me a lot when same command is implemented completely differently  :)

ShikOfTheRa, 2018-06-01:
ah - crossed over then. I will test at first opportunity. Thanks

ShikOfTheRa, 2018-06-08:
Looks good on the test bench. Thanks @stronnag 

Closing!

PR #3307 (2018-06-01): change year base for RTC time_t to correct leap processing
Modify the base year for RTC time_t calculation in order for the"modulo 4" calculations to work correctly for leap years. The base is 2000 (a leap year) and an offset added / subtracted to get back to the UNIX epoch.

Fixes #3302 (for issue for detail)

mwp log (day is now correct):
```
2018-06-01T08:12:40+0100 RTC local 2018-06-01T08:12:40.769, fc 2018-06-01T08:12:45.520
```
BB log, file system time and log remain correct
```
H Log start datetime:2018-06-01T07:20:02.762+00:00
```
closed 2018-06-08 by ShikOfTheRa